OPCFoundation/UA-.NETStandard #1313: How to change generated certificate expiry date

lasharn, 2021-03-07:
**Type of Issue**
[ ] Bug [ ] Enhancement [ ] Compliance [X ] Question [ ] Help wanted

Hi,

The certificate that is automatically generated by the reference server expires after 1 year. Is there a way to change the creation of the certificate so that I can change the expiry date?

Alternatively, if I create a certificate using Opc.Ua.CertificateGenerator.exe that has a new expiry date, how can I get my server to use that certificate instead of generating a new one?

I've looked at #678 which solves checking and renewing the certificate after its expired but it doesn't seem to solve creating a certificate with a longer expiry date.

Thanks

MD-V, 2021-03-08:
Method ApplicationInstance.CheckApplicationInstanceCertificate accepts lifeTimeInMonths parameter. If cert is created this lifetime is used for the cert.

See:   https://github.com/OPCFoundation/UA-.NETStandard/blob/4a14df9cd5372397cf51977b53a560f3b4d341eb/Libraries/Opc.Ua.Configuration/ApplicationInstance.cs#L317

lasharn, 2021-03-15:
Thanks, this solves my problem.

closed 2021-03-15 by lasharn