OPCFoundation/UA-.NETStandard #1318: Client Session Create static method is thread blocking while waiting for the server to accept or refuse the client certificate

PierluigiCarusoOcima, 2021-03-11:
**Type of Issue**
[x] Bug [ ] Enhancement [ ] Compliance [ ] Question [ ] Help wanted

**Describe the Issue**

The method Session.Create returns a Task and behaves asynchronously while establishing a connection to a server, but if the server has yet to decide wheter to trust or reject the client certificate the method will be synchronous.
This in particular causes issues when the server requires user input to make the decision (Like the one in the [samples](https://github.com/OPCFoundation/UA-.NETStandard-Samples))

**To Reproduce**
Steps to reproduce the behavior:
1. Compile Sample 

			ApplicationInstance application = new ApplicationInstance
			{
				ApplicationType = ApplicationType.Client
			};

			var configuration = application.LoadApplicationConfiguration("OPC-UA.config.xml", false).Result;

			application.CheckApplicationInstanceCertificate(true, 0, 1200).Wait();

			configuration.CertificateValidator.CertificateValidation += (sender, e) =>
			{
				e.Accept = true;
				if (configuration.SecurityConfiguration.TrustedPeerCertificates.OpenStore().FindByThumbprint(e.Certificate.Thumbprint).Result.Count > 0)
				{
					return;
				}

				configuration.SecurityConfiguration.RejectedCertificateStore.OpenStore().Delete(e.Certificate.Thumbprint);
				configuration.SecurityConfiguration.TrustedPeerCertificates.OpenStore().Add(e.Certificate);
			};

			// load the application configuration.
			var endpointDescription = CoreClientUtils.SelectEndpoint("opc.tcp://desktop-pd5e3ft:51210/UA/SampleServer", false, 5000);

			ConfiguredEndpoint endpoint = new ConfiguredEndpoint(null, endpointDescription,
				EndpointConfiguration.Create(configuration));

			var genericTask = Task.Delay(2000).ContinueWith(antecent => Console.WriteLine("Delayed task ended"));

			var sessionTask = Session.Create(
				configuration,
				endpoint,
				false,
				false,
				configuration.ApplicationName,
				1800000,
				null,
				null)
				.ContinueWith(antecent => Console.WriteLine("Create session ended"));

			Console.WriteLine("Created tasks");

			sessionTask.Wait();
			genericTask.Wait();`

2. Connect to Sample Server.NET4
3. Do not accept or refuse the certificate

**Expected behavior**
Session.Create should run asynchronously while waiting for the server to handle the client certificate.

**Screenshots**
Running the above code will produce this situation:
[image]

In which Session.Created has blocked the execution of the next line for a significant amount of time

After we instruct the server on how ot handle the certificate the program seems to work as expected
[image]

**Environment (please complete the following information):**
 - Runtime: .NET Core 3.1
 - OPC Package Release Version 1.4.365.48
 - Component: Opc.Ua.Core
 - Server: Sample Server

mregen, 2021-09-14:
Hi @Ocima, the behavior you describe is by design. Even considered an async version it has to wait for client confirmation to return the session object. 
I *think* what's missing here is really that the API signature for the client call should support a CancellationToken. 
The async client API will be improved over time due to the availability of new async stubs in core, please stay tuned. see #1498.

closed 2021-09-14 by mregen

PierluigiCarusoOcima, 2021-09-23:
The problem here is that the method is waiting for the client confirmation to return the Task<Session> rather than the session object itself