OPCFoundation/UA-.NETStandard #1319: Client read using a different encoding

kimaluco17, 2021-03-11:
Hello,

I am wondering if the client SDK supports XML or JSON encodings. If so, is there a way to specify which encoding to use?

AlinMoldovean, 2021-03-15:
Hello @kimaluco17 ,

The Client SDK has support for XML encoding.
It can be specified in the read requests like in the example below.
Please note that it only applies if the DataType of the Variable is a subtype of Structure.

```

                // build a list of nodes to be read
                ReadValueIdCollection nodesToRead = new ReadValueIdCollection()
                {
                    // Value of ServerStatus
                    new ReadValueId() { NodeId = Variables.Server_ServerStatus, AttributeId = Attributes.Value , DataEncoding = BrowseNames.DefaultXml },
                    // BrowseName of ServerStatus_StartTime
                    new ReadValueId() { NodeId = Variables.Server_ServerStatus_StartTime, AttributeId = Attributes.BrowseName},
                    // Value of ServerStatus_StartTime
                    new ReadValueId() { NodeId = Variables.Server_ServerStatus_StartTime, AttributeId = Attributes.Value,}
                };

                // Read the node attributes
                m_output.WriteLine("Reading nodes...");

                // Call Read Service
                m_session.Read(
                    null,
                    0,
                    TimestampsToReturn.Both,
                    nodesToRead,
                    out DataValueCollection resultsValues,
                    out DiagnosticInfoCollection diagnosticInfos);

```

kimaluco17, 2021-03-17:
Thanks as always @AlinMoldovean, that works.

closed 2021-03-17 by kimaluco17