OPCFoundation/UA-.NETStandard #1320: GetDefaultHistoryCapabilities does not return the existing node from address space

ososnilknarf, 2021-03-12:
**Type of Issue**
[ x] Bug [ ] Enhancement [ ] Compliance [ ] Question [ ] Help wanted

The method DiagnosticsNodeManager.GetDefaultHistoryCapabilities() does not return the existing HistoryCapabilities node from the address space. There is already one there because it is part of the predefined node model for that NodeManager. 
GetDefaultHistoryCapabilities() keeps a private member m_historyCapabilities. That private member is only ever initialized in that method, and it never attempts to find the existing instance in the address space.
Additionally, the HistoryServerCapabilities node that already exists in the address space BaseObjectState, not the derived class HistoryServerCapabilitiesState.
Also when HistoryServerCapabilities is created in GetDefaultHistoryCapabilities() it doesn't use the node ID constant that is provided with the OPC library, (Opc.Ua.ObjectIds.HistoryServerCapabilities), so the capabilities object would not be accessible using this nodeId.

What seems to be missing is to convert this node to HistoryServerCapabilitiesState in AddBehaviourToPredefinedNode(), and initialize m_historyCapabilities with that instance when it is created.

The following workaround seems to work to get the existing predefined node and update it:

```
                var nodeHistoryServerCapabilities = FindNodeInAddressSpace(Opc.Ua.ObjectIds.HistoryServerCapabilities);

                if (nodeHistoryServerCapabilities == null)
                    return;

                var qualAHDCName = new QualifiedName("AccessHistoryDataCapability");

                nodeHistoryServerCapabilities.SetChildValue(SystemContext, qualAHDCName, true, true);

```
This is not ideal though, since the HistoryServerCapabilitiesState class isn't able to be used.

**To Reproduce**
Steps to reproduce the behavior:
1. Add the following code to any NodeManager (this code is from the sample HistoricalAccess Server:
```
lock (Server.DiagnosticsLock)
{
    HistoryServerCapabilitiesState capabilities = Server.DiagnosticsNodeManager.GetDefaultHistoryCapabilities();
    capabilities.AccessHistoryDataCapability.Value = true;
    capabilities.InsertDataCapability.Value = true;
    capabilities.ReplaceDataCapability.Value = true;
    capabilities.UpdateDataCapability.Value = true;
    capabilities.DeleteRawCapability.Value = true;
    capabilities.DeleteAtTimeCapability.Value = true;
    capabilities.InsertAnnotationCapability.Value = true;
}
```
2. Connect to the Server
3. Browse Address space to the path "Objects/Server/ServerCapabilities/" and browse the children of that node
4. Observe that there are 2 instances of the HistoryServerCapabilities node. One of them contains the values that were set in Step 1.
5. Observe that the one with uninitialized values has the node id constant value Opc.Ua.ObjectIds.HistoryServerCapabilities

**Expected behavior**
There should be a single instance of the HistoryServerCapabilities node, containing the specified values, and using the proper node id, Opc.Ua.ObjectIds.HistoryServerCapabilities.

**Environment (please complete the following information):**
 - OPC Package Release Version [1.4.365.23]
 - Component: [e.g. Opc.Ua.Core]

AlinMoldovean, 2021-03-22:
Hello @ososnilknarf ,

Thanks for reporting this.
A fix was added in #1334

closed 2021-03-23 by AlinMoldovean

ososnilknarf, 2021-03-24:
Thanks Alin. 
When should this be available in the nuget package?

AlinMoldovean, 2021-03-25:
@ososnilknarf,

For the moment we do not have a fixed date for a NuGet package release.

PR #1334 (2021-03-23): Fix for duplicated HistoryServerCapabilities.
Fix for duplicated HistoryServerCapabilities node in DiagnosticsNodeManager. Fixes #1320.
commit 7e501ff4 (2021-03-23): Fix for duplicated HistoryServerCapabilities node in DiagnosticsNodeManager. Fixes #1320. (#1334)