OPCFoundation/UA-.NETStandard #1321: OPC Server and Notification

Giuseppe-1977, 2021-03-14:
Hi all,
I am coding an OPC client which wants to read a notification item and when It is happenend the client reads the new value and continues with the proper Logic. When the value of this item Is changed the OPC Ua server  sends multiple and identical notification message to the client. Why ? Don't  I receive One message ? Are there ant configuration settigs to be applied ?

AlinMoldovean, 2021-03-15:
Hi @Giuseppe-1977,

An example of subscribing to data change notifications can be found in ConsoleReferenceClient:

https://github.com/OPCFoundation/UA-.NETStandard/blob/d7236cebc1860a55fea9be7612381755e8656bbb/Applications/ConsoleReferenceClient/UAClient.cs#L373

Does the issue replicate using this example?

Giuseppe-1977, 2021-03-15:
The example Is fine, but when you have a thread which Is listenining the notification message you get Always the same change notification even of the tag value does not change anymore after the first time. The problem Is the OPC Ua server which I got bythe library. Do I Need to config properly?

AlinMoldovean, 2021-03-16:
@Giuseppe-1977 ,

In this case the problem might on the server side.
You can try to set a filter with `DataChangeTrigger.StatusValue` to avoid notifications when only timestamp changes, but this should be solved on the server side.

```
        // Create MonitoredItems for data changes

        MonitoredItem intMonitoredItem = new MonitoredItem(subscription.DefaultItem);
        // Int32 Node - Objects\CTT\Scalar\Simulation\Int32
        intMonitoredItem.StartNodeId = new NodeId("ns=2;s=Scalar_Simulation_Int32");
        intMonitoredItem.AttributeId = Attributes.Value;
        intMonitoredItem.DisplayName = "Int32 Variable";
        intMonitoredItem.SamplingInterval = 1000;
        intMonitoredItem.Notification += OnMonitoredItemNotification;

        // Set a DataChangeFilter for the item.
        DataChangeFilter monitoringFilter = new DataChangeFilter();
        monitoringFilter.Trigger = DataChangeTrigger.StatusValue;
        intMonitoredItem.Filter = monitoringFilter;
```

Giuseppe-1977, 2021-03-23:
Thank, but all this is not working fine because my client continues to
receive the same notification. All below is inside into my dispatcher
thread.

   private void SubscriptionForAll()
        {
            string tagToRead = "";
            try
            {
                if (mySubscription == null)
                {
                    mySubscription = uaClient.Subscribe(0);
               }
                monitoredItemNameContapezzi = "ns=2;s=ContaPezzi";
                tagToRead = "ns=2;s=ContaPezzi_Instructions";

                // Create MonitoredItems for data changes
                MonitoredItem intMonitoredItem = new
MonitoredItem(mySubscription.DefaultItem);

                // Int32 Node - Objects\CTT\Scalar\Simulation\Int32
                intMonitoredItem.StartNodeId = new NodeId(tagToRead);
                intMonitoredItem.AttributeId = Attributes.Value;
                intMonitoredItem.DisplayName = "ContaPezzi";
                intMonitoredItem.SamplingInterval = 0;
                intMonitoredItem.Notification += Notification_MonitoredItem;

                DataChangeFilter monitoringFilter = new DataChangeFilter();
                monitoringFilter.Trigger = DataChangeTrigger.StatusValue;
                intMonitoredItem.Filter = monitoringFilter;

               mySubscription.AddItem(intMonitoredItem);
                mySubscription.ApplyChanges();

            }
            catch (Exception ex)
            {
                MessageBox.Show(ex.Message, "Error");
            }

        }

   private void Notification_MonitoredItem(MonitoredItem monitoredItem,
MonitoredItemNotificationEventArgs e)
        {
            MonitoredItemNotification notification = e.NotificationValue as
MonitoredItemNotification;
            if (notification == null) return;
            if (monitoredItem.DisplayName == "ContaPezzi")
            {
                if (Convert.ToInt32(notification.Value.WrappedValue.Value)
== 0) return;
                _CountWorkDone = _CountWorkDone +
Convert.ToDecimal(notification.Value.WrappedValue.Value);
                Lavorazione.SetWorkDone(_CountWorkDone);

bridgeServices.UpdateCountProduction(Lavorazione.LavorazioneId,
_CountWorkDone);
            }
        }

-- 
* CAPUANO GIIUSEPPE*
Via Privata le Primule, 3 - 23818 Pasturo (LC)
Tel. 3403901097

RISERVATEZZA ​​Nel rispetto del Codice Privacy (D.Lgs 196/03) sulla tutela
dei dati personali e dell'art. 616 c. p. questo messaggio e i contenuti dei
suoi allegati sono riservati e confidenziali ed esclusivamente indirizzati
al destinatario che può farne uso per le finalità indicate nel messaggio.
Se non siete i destinatari del messaggio o lo avete ricevuto per errore, Vi
preghiamo di dare comunicazione al mittente al più presto e di rimuovere il
messaggio e gli allegati dal Vostro sistema. I contenuti di questo
messaggio e degli allegati non devono essere usati, copiati o diffusi in
alcun modo

AlinMoldovean, 2021-03-24:
@Giuseppe-1977 ,

I suspect that in your case the UA Server is generating unnecessary DataChange Notifications.
In case your server is developed with this library, could you provide the steps to reproduce the issue?
When testing with _Reference Server_ from this repository the server does not send multiple notifications with the same value for static nodes. e.g. node _ns=2;s=Scalar_Static_Int32_ located at _Objects\CTT\Scalar\Scalar_Static\Int32_

AlinMoldovean, 2021-04-06:
@Giuseppe-1977 ,

please reopen if you can provide the steps to reproduce the issue when testing against _ReferenceServer_.

closed 2021-04-06 by AlinMoldovean

Giuseppe-1977, 2021-05-17:
> Does the issue replicate using this sample code?

Dear Sir,

How can I connect by using User and Password?

      SetStatusMessage("Inizializzazione Connessione PLC Bridge in corso
...");
                            uaClientBridge = new
UAClientHelperAPI(application.ApplicationConfiguration, console,
ClientBase.ValidateResponse);
                            uaClientBridge.ServerUrl =
config.GetSection("PLC:Bridge:IP").Value;

                            //myClientHelperAPI.Connect(mySelectedEndpoint,
userPwButton.Checked, userTextBox.Text, pwTextBox.Text);

                            connectBridge = await
uaClientBridge.ConnectAsync();
                            mySessionBridge = uaClientBridge.Session;
                            uaClientBridge.KeepAliveNotification += new
KeepAliveEventHandler(Notification_KeepAlive);

-- 
* CAPUANO GIIUSEPPE*
Via Privata le Primule, 3 - 23818 Pasturo (LC)
Tel. 3403901097

RISERVATEZZA ​​Nel rispetto del Codice Privacy (D.Lgs 196/03) sulla tutela
dei dati personali e dell'art. 616 c. p. questo messaggio e i contenuti dei
suoi allegati sono riservati e confidenziali ed esclusivamente indirizzati
al destinatario che può farne uso per le finalità indicate nel messaggio.
Se non siete i destinatari del messaggio o lo avete ricevuto per errore, Vi
preghiamo di dare comunicazione al mittente al più presto e di rimuovere il
messaggio e gli allegati dal Vostro sistema. I contenuti di questo
messaggio e degli allegati non devono essere usati, copiati o diffusi in
alcun modo

AlinMoldovean, 2021-05-18:
> 
> Dear Sir, How can I connect by using User and Password?

 - By sending the User and Password in `UserIdentity` at `Session.Create()`:

```
                    // Create the session
                    Session session = await Session.Create(
                        m_configuration,
                        endpoint,
                        false,
                        false,
                        m_configuration.ApplicationName,
                        30 * 60 * 1000,
                        new UserIdentity("userName", "password"),
                        null
                    );

```