OPCFoundation/UA-.NETStandard #1323: Empty UserToken SecurityPolicy not handled properly

ToSa27, 2021-03-15:
**Type of Issue**
[ X ] Bug [ ] Enhancement [ ] Compliance [ ] Question [ ] Help wanted

**Describe the Issue**
The OPC UA specification states in 7.36.2.1 that it is a valid option to not provide a SecurityProfile as part of the UserIdentityToken and in such case the SecurityProfile of the EndpointDescription should be used instead.
In OPC.UA.Client if the EndpointDescription SecurityPolicy is set to None and the UserIdentityToken is set to a null string, the code still jumps into certificate validation even if it should not.

**To Reproduce**
Steps to reproduce the behavior:
1. Compile Sample (any sample client should work)
2. Connect to any Server that only supports None and provides an empty (null) UserIdentityToken SecurityPolicy with an untrusted certificate
3. See error: session can't be established because certificate is untrusted - but it should create the session because the SecurityPolicy None should be used taken from the EndpointDescription

**Expected behavior**
Instead the connection should be established not even checking the certificates.

**Environment (please complete the following information):**
 - OS: Windows 10
 - Development environment: Visual Studio 2019
 - Runtime: .NET 4.6.2 and .NET Core 3.1
 - OPC Package Release Version: 1.4.365
 - Component: Opc.Ua.Client
 - Server: Siemens PLCSIM Advanced
 - Client: build based on OPC.Ua.Client

ToSa27, 2021-03-15:
From the specification (release 1.04):
If this SecurityPolicy is omitted then the Client uses the SecurityPolicy in the EndpointDescription. If the matching SecurityPolicy is set to None then no encryption or signature is required.

Added a pull request to fix this: #1324

AlinMoldovean, 2021-03-16:
Hello @ToSa27 ,

Thanks for pointing this out. This was already solved on _UA-.NET-Legacy_ repository [here ](https://github.com/OPCFoundation/UA-.NET-Legacy/commit/9b0a2edf7c9c671808af6e4f589f4a3934e96052#diff-f4745cf95209f646909cf3b8c9258d40aaf3165bbc74499a246deb11563cbfff) but we missed this place.
Your fix has minor flaw that does not check for empty strings also, but I'll put a note on PR for this.

Thanks,
Alin

closed 2021-03-23 by AlinMoldovean