OPCFoundation/UA-.NETStandard #1327: Getting DisplayName of Nodes in FastDataChangeNotificationEventHandler

auxua, 2021-03-18:
Hi everyone,

Currently I am shifting the DataAcquisition to Subscriptions. As there are multiple Values I want to observe, I create one Subscription with a set of MonitoredItems and use the FastDataChangeNotificationEventHandler. On Data changes the delegate/method gets called and I can get the new values without problems. But I cannot see to what Node the Value belongs to. How can I get that information?

The signature of the handler is
```csharp
public static void FastHandler(Subscription subscription, 
            DataChangeNotification notification, IList<string> stringTable)
```
I find the updated value in `notification.MonitoredItems.ElementAt(i).Value.Value`, but in the element (Type `MonitoredItemNotification`) there is no DisplayName or Node ID allowing me to identify the Node, the value belongs to.

The parameter `subscription` holds information about the actual subscription, but does not tell, which nodes/items in it are updated.

So for example: 
I am monitoring the Nodes A, B and C.
A and C change their values
I get a Call to the FastHandler as expected
In `notification.MonitoredItems` I find 2 Items, but no indication, what nodes there are linking to.

Am I missing something in here? How can I link those new values to the Nodes I am looking into?

AlinMoldovean, 2021-03-18:
Hi @auxua ,

You need to use `MonitoredItemNotification.ClientHandle`  to identify the `MonitoredItem `inside `Subscription.MonitoredItems` list.

auxua, 2021-03-18:
> Hi @auxua ,
> 
> You need to use `MonitoredItemNotification.ClientHandle` to identify the `MonitoredItem `inside `Subscription.MonitoredItems` list.

Wow... I was searching to many fields and properties mitaking ClientHandle for a handle to the opc client. So solved my hours of searching with a one-liner - Thank you very much! =D

closed 2021-03-18 by auxua