OPCFoundation/UA-.NETStandard #1328: Request to get access to azure tenant opcfoundation-prototyping

charlesaugustineabb, 2021-03-18:
**Type of Issue**
[ ] Bug [ ] Enhancement [ ] Compliance [ ] Question [ ] Help wanted
Help wanted

**Describe the Issue**

https://github.com/OPCFoundation/UA-.NETStandard/tree/prototyping-OAuth2

I am trying to execute sample projects( UserAuthentication Client and UserAuthentication Server)  provided in above path.
to understand  about  jwt tokens in opc ua session.
UserAuthentication Client  asks to login to azure identity server for getting the tokens. since my azure login (charles.augustine@in.abb.com) dont have gets the below error.
AADSTS50020: User account 'charles.augustine@in.abb.com' from identity provider 'https://sts.windows.net/372ee9e0-9ce0-4033-a64a-c07073a91ecd/' does not exist in tenant 'opcfoundation-prototyping' and cannot access the application 'f8b31779-d9a3-43ff-b854-28f27a52e2f2'(prototype-client) in that tenant. The account needs to be added as an external user in the tenant first. Sign out and sign in again with a different Azure Active Directory user account.

Is there a possibility to get access permission for me to access azure tenant opcfoundation-prototyping ? So i can explore the sample  
my email id is charles.augustine@in.abb.com
or any other suggestions?

**To Reproduce**
Steps to reproduce the behavior:
1. Compile Sample '...'
2. Connect to Server '....'
3. Browse Address space '....'
3. Subscribe to Node '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Log output**
If applicable, add log output to help explain your problem.

**Code snippets**
If applicable, add a code snippet to help explain your problem.

**Environment (please complete the following information):**
 - OS: [e.g. Windows 10, Ubuntu, macOS]
 - Development environment: [e.g. Visual Studio 2019 16.7.2]
 - Runtime: [e.g. .NET 4.6.2, .NET Core 3.1]
 - OPC Package Release Version [e.g. 1.4.363.107]
 - Component: [e.g. Opc.Ua.Core]
 - Server: [e.g. Reference Server]
 - Client: [e.g. UA Expert]

**Additional context**
Add any other context about the problem here.

AlinMoldovean, 2021-03-22:
Hello @charlesaugustineabb ,

The prototype provides an example of using an Azure OAuth2 service for validating user tokens.
The Azure Active Directory account was just a temporary account used for testing and it is not intended for public access.
You need to configure a separate Azure subscription for further testing this sample.

charlesaugustineabb, 2021-03-23:
Thanks for clarification.

closed 2021-03-23 by charlesaugustineabb