OPCFoundation/UA-.NETStandard #1329: Subscribing to a non-existing node does not provide feedback to indicate the node's status

hermoter, 2021-03-19:
When subscribing to a non-existing node, the subscription is created and everything appears to be successful,
except that you won't receive any notifications for this particular monitored item.

The server provides no feedback telling the client that the node id does not exist.
This could lead to a situation where a client may think the subscription was successful, but may be waiting indefinitely to receive notifications, not knowing that the lack of notifications is because of the node not existing.

Suggestions I have found is to read a node's attribute to check if it exists, but the issue with this is that it is very expensive, especially when subscribing to thousands of nodes at once. It overloads the OPC UA server unnecessarily.

When calling subscription.Create(), why does it return a status of Good, even though there are monitored items for non-existent nodes?

Could the Create() method not return a list of status codes for each monitored item, similar to the Read command of a list of node ids?

AlinMoldovean, 2021-03-22:
Hello @hermoter ,

In your case the Subscription was created successfully on server side, but some monitored were not created due to some errors.
You can get the status of the MonitoredItems of a Subscription in this way:

```

          // Create the monitored items on Server side
          subscription.ApplyChanges();
          m_output.WriteLine("MonitoredItems created for SubscriptionId = {0}.", subscription.Id);

          foreach (MonitoredItem item in subscription.MonitoredItems)
          {
              bool createdOnserver = item.Created;
              ServiceResult itemError = item.Status.Error;
          }
```

[image]

hermoter, 2021-03-23:
Thank you very much for the clarification @AlinMoldovean!
I will give this a try this week.

closed 2021-03-31 by AlinMoldovean