OPCFoundation/UA-.NETStandard #1333: Async problem in class Opc.Ua.NodeIdDictionary, freezing Reference Client

CKa77, 2021-03-20:
**Type of Issue**
[X] Bug [ ] Enhancement [ ] Compliance [ ] Question [ ] Help wanted

**Describe the Issue**

The reference C# client was most of the times I connected to Integration Object's free OPC UA Server Simulator freezing and not showing the nodes and tags in the browser window. Sometimes, but rarely it worked, and when trying to restart the reference client it would just freeze after connecting.

**To Reproduce**
Steps to reproduce the behavior:
1. Compile Reference Client and start it from Visual Studio Debugger (press F5)
2. Download, install and start OPC Foundation's OPC UA Server Simulator
3. Connect with the Reference Client to the OPC UA Server Simulator address
4. Most of the time, the Reference Client's UI will freeze completely

**Expected behavior**
Most of the times, the Reference Client's UI will freeze

**Code snippets**
I managed to get the Reference Client to work reliably, by introducing lock-statements for the variable 'm_numericIds' in class Opc.Ua.NodeId, in methods Add(NodeId, T value) and TryGetValue(NodeId key, out T value), as follows:

In Add() method, the following case-statement was changed as follows:
case IdType.Numeric:
                {
                    ulong id = ((ulong)key.NamespaceIndex) << 32;
                    id += (uint)key.Identifier;

                    lock (m_numericIds)
                    {
                        m_numericIds.Add(id, value);
                    }

                    return;
                }

And in TryGetValue method, the following code was used:
case IdType.Numeric:
                {

                    ulong id = ((ulong)key.NamespaceIndex) << 32;
                    id += (uint)key.Identifier;

                    bool _result = false;

                    lock (m_numericIds)
                    {
                       _result = m_numericIds.TryGetValue(id, out value);
                    }

                    return _result;

                    //return m_numericIds.TryGetValue(id, out value);

**Environment (please complete the following information):**
 - OS: Windows 10 Pro
 - Development environment: Visual Studio 2019 16.9.2
 - Runtime: NET 4.6.2
 - OPC Package Release Version: Latest from GitHub master.
 - Component: [e.g. Opc.Ua.NodeIdDictionary
 - Server: Integrated Object's OPC UA Server Simulator
 - Client: OPC Foundation's Reference Client

**Additional context**
After the changes shown above, the Reference Client was working. I do not say that this is necessary the absolute correct fix here, but I think this shows where the problem is. A fix to the master copy would be good to implement.

mrsuciu, 2021-09-15:
Needs to be checked

mregen, 2024-11-29:
The implementation of the NodeIdDictionary has been changed recently to have a normal Dictionary as underlying dictionary. If there is still a contention issue needs to be checked. An alternative were also to use a ConcurrentDictionary. please reopen if there is still an issue.

closed 2024-11-29 by mregen