OPCFoundation/UA-.NETStandard #1339: jwt(JSON Web Token) support for Dotnet SDK OPCFoundation.NetStandard.Opc.Ua

charlesaugustineabb, 2021-03-26:
**Type of Issue**
[ ] Bug [ ] Enhancement [ ] Compliance [X] Question [ ] Help wanted
Question 

**Describe the Issue**

Dotnet SDK OPCFoundation.NetStandard.Opc.Ua supports jwt(JSON Web Token) ?
if not, Kindly let me know, when SDK with  jwt support planning to release? at least a tentative plan?

A clear and concise description of what the issue is.

**To Reproduce**
Steps to reproduce the behavior:
1. Compile Sample '...'
2. Connect to Server '....'
3. Browse Address space '....'
3. Subscribe to Node '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Log output**
If applicable, add log output to help explain your problem.

**Code snippets**
If applicable, add a code snippet to help explain your problem.

**Environment (please complete the following information):**
 - OS: [e.g. Windows 10, Ubuntu, macOS]
 - Development environment: [e.g. Visual Studio 2019 16.7.2]
 - Runtime: [e.g. .NET 4.6.2, .NET Core 3.1]
 - OPC Package Release Version [e.g. 1.4.363.107]
 - Component: [e.g. Opc.Ua.Core]
 - Server: [e.g. Reference Server]
 - Client: [e.g. UA Expert]

**Additional context**
Add any other context about the problem here.

charlesaugustineabb, 2021-04-01:
i could make it work using IssuedIdentityToken .

AlinMoldovean, 2021-04-01:
@charlesaugustineabb ,

The SDK should be able to handle a JWT user identity inside a `IssuedIdentityToken` using the `byte[] TokenData` field.
The existing `prototype ` provided an example of handling tokens issued by in _Azure AD_

closed 2021-04-14 by AlinMoldovean