OPCFoundation/UA-.NETStandard #1355: Enable Discussion feature

MD-V, 2021-04-08:
**Type of Issue**
- [ ] Bug 
- [X] Enhancement 
- [ ] Compliance 
- [ ] Question 
- [ ] Help wanted

**Describe the Issue**

It would be neat to enable GitHubs discussion feature for this repo. This would lead to a better distinction between issues and questions/discussions. What are your thoughts on this?

**To Reproduce**
-

**Expected behavior**
-

**Screenshots**
-

**Log output**
-

**Code snippets**
-

**Environment (please complete the following information):**
 -

**Additional context**
-

mregen, 2021-05-03:
Hi @MD-V , there has been a question by the OPCF admin on what value the Discussions feature adds compared to Issues beside having better distinction? Could you elaborate?

mregen, 2021-06-01:
request was refused by OPCF

closed 2021-06-01 by mregen

mregen, 2023-07-03:
Discussions are enabled!