OPCFoundation/UA-.NETStandard #1356: [Help wanted] Subscription missing values

Lmpas28, 2021-04-08:
Hi, I'm having some issues with subscriptions but I'm not sure if is a client side or server side problem so I'm posting here to find out if I'm doing things correctly or not. 

I'm using this nuget package to develop an opc-ua client. As of now everything worked as expected but I have a new opc-ua server that I need to connect to and after some tests I've seen that I'm missing data. The server has almost 2k variables (all changing) and I'm using a subscription to get all those data changes. As I would like to receive all data changes in one call I'm using the fast data change callback.

Here is how I set up the subscription:

		private void ActivateSubscription()
		{
			// Activate subscription.
			try
			{
				_connHelper.GenerateSubscription(_config);
				// Subscribe to notifications
				//_connHelper.Notification += OnNotification;
				_config.OpcSubscription.FastDataChangeCallback = OnDataChange;
			}
			catch (Exception err)
			{
				_logger.Error("An error occurred while generating the subscription: " + err.ToString());
			}
		}

here is the callback:

		private void OnDataChange(Subscription subscription, DataChangeNotification notification, IList<string> stringTable)
		{
			_logger.Info("Number of items received from opcua -> " + notification.MonitoredItems.Count);
			foreach (MonitoredItemNotification itemNotification in notification.MonitoredItems)
			{
				MonitoredItem monitoredItem = subscription.FindItemByClientHandle(itemNotification.ClientHandle);

				if (monitoredItem == null)
				{
					continue;
				}
				else
				{
                                        // custom code to process the monitored item
				}
                        }
               }

In the server ui I can see my subscription created and I'm receiving notifications in my callback so it is correctly configured. In my tests I've checked that apparently the server is only capable of sending blocks of max 100 variables in each publish request, so no matter If I have 500 or 1000 variables in the server or If I set a maxnotificationLimit above 100, it sends blocks of 100. So If I had 500 variables I expect to receive 5 blocks of 100 variables in the fastdatachange callback. I've done several tests changing sampling and publishing intervals and I'm always missing data when having more than ~ 500 variables monitored. 

The thing is that Im not sure if having the fastdatachange callback enabled is enough from my side and the library takes care of the rest or If I should be doing manual republish requests when I see missing messages. Or maybe the problem is in the server that is not able to publish all data changes when it has a "high" amount of variables ?.

I'm also seeing jumps in the sequence numbers received in the callback:
[image]

Any help is appreciated! thanks!

AlinMoldovean, 2021-04-09:
Hi @Lmpas28 ,

`maxNotificationsPerPublish `represents "_The maximum number of notifications that the Client wishes to receive in a single Publish response_".
In some cases the server is not able to send as many notifications as requested, but anyway, in that case it should send all data in multiple Publish responses.
Your code looks ok, but maybe you are dealing with an server with an uncompliant behavior.
I can suggest to split the monitored items in multiple subscriptions and check the results. e.g. only 100 Monitoreditems per subscription.

Lmpas28, 2021-04-13:
> Hi @Lmpas28 ,
>
> `maxNotificationsPerPublish `represents "_The maximum number of notifications that the Client wishes to receive in a single Publish response_".
> In some cases the server is not able to [...]

Thanks @AlinMoldovean, I will try that because apparently the server is not able to publish all changes when having only one subscription with all the variables.

vinaybr, 2021-04-21:
Do you have an update on this issue. I am facing a similar issue.

mrsuciu, 2021-09-15:
Is this still an issue ? Please re-open if this is still an issue .

closed 2021-09-15 by mrsuciu