OPCFoundation/UA-.NETStandard #1361: SHA1 signed certificates are not trusted

enricobenedos, 2021-04-14:
**Type of Issue**
- [ ] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [ ] Question 
- [x] Help wanted

**Describe the Issue**

Good morning,

I'm trying to connect to a OPC UA server that expose some specific tag only using username and password, but when my net core console app try to create the session this exception is generated:
```
Opc.Ua.ServiceResultException: 'SHA1 signed certificates are not trusted.'
```

**To Reproduce**
```c#
var appconfig = new ApplicationConfiguration()
            {
                ApplicationName = "Test Client",
                ApplicationType = ApplicationType.Client,
                SecurityConfiguration = new SecurityConfiguration
                {
                    ApplicationCertificate = new CertificateIdentifier
                    {
                        StoreType = "X509Store",
                        StorePath = "CurrentUser\\My",
                        SubjectName = "UA Core Sample Client"
                    },
                    TrustedPeerCertificates = new CertificateTrustList
                    {
                        StoreType = "Directory",
                        StorePath = "OPC Foundation/CertificateStores/UA Applications",
                    },
                    TrustedIssuerCertificates = new CertificateTrustList
                    {
                        StoreType = "Directory",
                        StorePath = "OPC Foundation/CertificateStores/UA Certificate Authorities",
                    },
                    RejectedCertificateStore = new CertificateTrustList
                    {
                        StoreType = "Directory",
                        StorePath = "OPC Foundation/CertificateStores/RejectedCertificates",
                    },
                    NonceLength = 32,
                    AutoAcceptUntrustedCertificates = true,
                    RejectSHA1SignedCertificates = false,
                    MinimumCertificateKeySize = 1024,
                },
                TransportConfigurations = new TransportConfigurationCollection(),
                TransportQuotas = new TransportQuotas { OperationTimeout = 15000 },
                ClientConfiguration = new ClientConfiguration { DefaultSessionTimeout = 60000 },
            };

            // Get the endpoint by connecting to server's discovery endpoint.
            // Try to find the first endopint without security.
            EndpointDescription endpointDescription = CoreClientUtils.SelectEndpoint("opc.tcp://192.168.3.4/", false);

            EndpointConfiguration endpointConfiguration = EndpointConfiguration.Create(appconfig);
            ConfiguredEndpoint endpoint = new ConfiguredEndpoint(null, endpointDescription, endpointConfiguration);

            // Create the session
            Session session = await Session.Create(
                appconfig,
                endpoint,
                false,
                false,
                appconfig.ApplicationName,
                30 * 60 * 1000,
                new UserIdentity("username", "password"),
                null
            );

            // Assign the created session
            if (session != null && session.Connected)
            {
                // Session created successfully.
                Console.WriteLine($"New Session Created with SessionName = {session.SessionName}");
            }
```

**Expected behavior**
Using `RejectSHA1SignedCertificates = false,` I expect to not receive any error. 

Anyway my server expose Basic256Sha256 but I'm not able to understand how to force my code to use it.

**Environment (please complete the following information):**
 - OS: Windows 10
 - Development environment: Visual Studio 2019 16.9.4
 - Runtime: .NET Core 3.1
 - OPC Package Release Version 1.4.365.48

AlinMoldovean, 2021-04-15:
Hi @enricobenedos ,

The error with that specific message should not be generated in case `RejectSHA1SignedCertificates = false`
You need to debug inside `CertificateValidator.InternalValidate()` and check this place:

```
            // check if minimum requirements are met
            if (m_rejectSHA1SignedCertificates && IsSHA1SignatureAlgorithm(certificate.SignatureAlgorithm))
            {
                sresult = new ServiceResult(StatusCodes.BadCertificatePolicyCheckFailed,
                    null, null, "SHA1 signed certificates are not trusted.", null, sresult);
            }
```

mregen, 2021-04-15:
Hi @enricobenedos , in order to Select security use SelectEndpoint (useSecurity = true);
If your server supports Basic256Sha256 it must have a SHA2 signed certificate.
Maybe you really see this issue: https://github.com/OPCFoundation/UA-.NETStandard/issues/1228
before the error can be suppressed.

enricobenedos, 2021-04-15:
Hi @AlinMoldovean,

thank you for your reply. Looking at the code you show me it seems that `RejectSHA1SignedCertificates` is not updated with the given value or something else is overriding it. 

Probably in my case when `CertificateValidator` is created `m_rejectSHA1SignedCertificates = CertificateFactory.DefaultHashSize >= 256;` should be true. But then the value should be updated by `m_rejectSHA1SignedCertificates = configuration.RejectSHA1SignedCertificates;`.

What is the better way to debug it? Can it be considered a bug?

@Hi @mregen,

you are right, the server is returning a SHA1 certificate instead of SHA256. We are trying to solve this problem with the OPC UA server device producer. The issue you linked me help me to bypass the problem subscribing to `CertificateValidation` event 😊.

But at this point we have two bugs: one related with `RejectSHA1SignedCertificates` and one related to `AutoAcceptUntrustedCertificates`. It seems that both the parameters/properties are ignored.

mregen, 2021-05-03:
Hi @enricobenedos, the SHA1 configuration is updated when you call `certificatevalidator.Update(securityConfiguration)`. Please check also #1386 which allows now also the app to manually set the flags.

closed 2021-05-19 by AlinMoldovean