OPCFoundation/UA-.NETStandard #1366: BadCertificateUseNotAllowed 'Usage of certificate is not allowed. exception on Self Signed User Certificate

CJCombrink, 2021-04-16:
**Type of Issue**
- [ ] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [x] Question 
- [x] Help wanted

**Describe the Issue**

When I try to connect to the Reference Server using a self signed user Certificate I get the following error from the server: `BadIdentityTokenInvalid` (see the log output below)

I did copy my user certificate ("john.der") file into both the "pki\trusted\certs" and "pki\trustedUser" folders. 
My application trust is set up already, I can connect using Anonymous and Username Password.

I used the following page as a reference to create the X509 User Certificates: https://documentation.unified-automation.com/uasdkhp/1.3.1/html/config_user_authentication.html#autotoc_md25

I am trying to connect from the UaExpert application and get the above error. 

Is this supposed to work, is there something wrong with creating the certificate or is it a configuration error?

**To Reproduce**
Steps to reproduce the behavior:
1. Generate the User Certificates
2. Run the Reference Server
3. Try to connect using the User Certificates
3. Error appears in the logs 

**Expected behavior**
I Expect the application to connect to the server without issue. 

**Log output**
```
2021-04-16 10:40:57.042 +02:00 [WRN] ***EXCEPTION*** BadCertificateUseNotAllowed 'Usage of certificate is not allowed.'
2021-04-16 10:40:57.074 +02:00 [WRN]  -- BadCertificateUseNotAllowed 'Usage of certificate is not allowed.'
2021-04-16 10:40:57.074 +02:00 [WRN]  -- BadCertificateUntrusted 'Certificate is not trusted.
SubjectName: E=john@internet.com, CN=john, OU=AB Company, O=Abcdef Bdefgh, L=Cape Town, S=Western Cape, C=ZA
IssuerName: E=john@internet.com, CN=john, OU=AB Company, O=Abcdef Bdefgh, L=Cape Town, S=Western Cape, C=ZA
'
2021-04-16 10:40:57.075 +02:00 [ERR] Certificate 'E=john@internet.com, CN=john, OU=AB Company, O=Abcdef Bdefgh, L=Cape Town, S=Western Cape, C=ZA' rejected. Reason=BadCertificateUseNotAllowed 'Usage of certificate is not allowed.'
2021-04-16 10:40:57.075 +02:00 [ERR] Writing rejected certificate to directory: [Directory]%LocalApplicationData%/OPC Foundation/pki/rejected
2021-04-16 10:40:57.105 +02:00 [WRN] ***EXCEPTION*** BadCertificateUseNotAllowed ServiceResultException Usage of certificate is not allowed.
2021-04-16 10:40:57.106 +02:00 [WRN] ***EXCEPTION*** BadIdentityTokenInvalid (http://opcfoundation.org/Quickstart/ReferenceServer/v1.04:InvalidCertificate) ''E=john@internet.com, CN=john, OU=AB Company, O=Abcdef Bdefgh, L=Cape Town, S=Western Cape, C=ZA' is an invalid user certificate.'
2021-04-16 10:40:57.108 +02:00 [WRN] ***EXCEPTION*** BadIdentityTokenInvalid (http://opcfoundation.org/Quickstart/ReferenceServer/v1.04:InvalidCertificate) 'InvalidCertificate'
```

**Code snippets**
Using the reference server as is.

**Environment (please complete the following information):**
 - OS: Windows 10
 - Development environment: Visual Studio 2019 16.9.2
 - Runtime: .NET 4.7.0
 - OPC Package Release Version: 1.4.365.48
 - Component: Opc.Ua.Core
 - Server: Reference Server
 - Client: UA Expert

**Additional context**
None

AlinMoldovean, 2021-04-16:
Hello @CJCombrink ,

The problem is that your user certificate does not have the _DataEncipherment_ flag set in _X509KeyUsageExtensions_
I think you can fix this with the following setting in your _openssl.cfg_:
`keyUsage = digitalSignature, nonRepudiation, keyEncipherment, dataEncipherment`

Another option is to implement a custom logic  for validating the user certificates in `ReferenceServer.VerifyUserTokenCertificate( )` 

For the moment ReferenceServer is passing the _"User X509"_ tests of UA Compliance Test Tool.
We need to check if _DataEncipherment_ flag should be mandatory for user certificates.

CJCombrink, 2021-04-16:
@AlinMoldovean 
Thank you for the information.

I created a new user certificate with the  `dataEncipherment` flag set. 
After I copied it to the "pki\trustedUser\certs" folder I could connect from my client to the Reference Server. 

FYI: I used this guide to create the user certificate: https://medium.com/coding-snippets/acting-as-a-certificate-authority-with-openssl-a0d71b17bc95

closed 2021-04-16 by CJCombrink