OPCFoundation/UA-.NETStandard #1368: Read raw history

dalekseev-monokot, 2021-04-19:
Hello everyone,

I can't figure out how to properly implement the function of historical raw reading. Should the values obtained as a result of historical update operations, such as after replacement, be returned as raw data?

Thanks.

AlinMoldovean, 2021-04-20:
Hello @dalekseev-monokot ,

You should have a look at the [Historical Access](https://github.com/OPCFoundation/UA-.NETStandard-Samples/tree/master/Workshop/HistoricalAccess) project from _UA-.NETStandard-Samples_ repository.
You can find an implementation example there.

dalekseev-monokot, 2021-04-20:
Hello @AlinMoldovean ,

I looked at the code example and based on the following methods ([this one](https://github.com/OPCFoundation/UA-.NETStandard-Samples/blob/8e25524a1357e321f9c8498ba0f8ae12818d3958/Workshop/HistoricalAccess/Server/UnderlyingSystem/ArchiveItemState.cs#L208) and [this one](https://github.com/OPCFoundation/UA-.NETStandard-Samples/blob/8e25524a1357e321f9c8498ba0f8ae12818d3958/Workshop/HistoricalAccess/Server/UnderlyingSystem/ArchiveItemState.cs#L584)) I conclude that a raw read should always include the modified data in the result, but I am confused that the specification defines the following (6.4.3.2 Read raw functionality):

`When this structure is used for reading Raw Values (isReadModified is set to FALSE), it reads the values, qualities, and timestamps from the history database for the specified time domain for one or more HistoricalDataNodes. This parameter is intended for use by a Client that wants the actual data saved within the historian`

...since the modified values are not the actual values saved by the historian.

closed 2021-05-21 by AlinMoldovean