OPCFoundation/UA-.NETStandard #1369: Problem with Security Policy Aes256_Sha256_RsaPss and UaCPPServer

ReIaxo, 2021-04-19:
Hi,

I recently added the option to use `SecurityPolicies.Aes128_Sha256_RsaOaep` and `SecurityPolicies.Aes256_Sha256_RsaPss` to our Client. The first one works like expected. The second causes this exception when calling Session.Create(...): 
> BadSecurityChecksFailed 'The chosen security policy is not supported by the client to connect to the server.

Is there something I need to configure in addtion? I set EndpointConfiguration.SecurityPolicyUri to `SecurityPolicies.Aes256_Sha256_RsaPss`.

In the UaCPPServer Demo Server (Version 1.7.4.520) I configured with the UA Admin Dialog, that SecurityPolicies  have to be `Aes256_Sha256_RsaPss` and SecurityMode = `Sign + Sign&Encrypt `.

We use the NuGet Package Version 1.4.365.48 for the Client.

Or is there an Issue with die Demo Server?

Thank You in advance.

mregen, 2021-04-20:
Hi @ReIaxo , the option with Pss padding is not supported on all platforms. It is determined at runtime.
Some Linux flavors may not support it.
The only way to find out is by checking `SecurityPolicies.GetUri` or `SecurityPolicies.GetDisplayNames` if it is listed.

ReIaxo, 2021-04-27:
Hi, thank you.
So you mean Client-Platform? Im on Windows 10 and `SecurityPolicies.GetDisplayNames` does not contain `SecurityPolicies.Aes256_Sha256_RsaPss`. Do you know if Windows10 should support `SecurityPolicies.Aes256_Sha256_RsaPss`?

mregen, 2021-04-28:
Hi @Relaxo, maybe there is an issue with the Pss detection. --> needs investigation. Windows 10 should fully support the Pss.

mregen, 2021-04-29:
works here -- which framework do you use? netcoreapp3_1 or netcoreapp2_1 or .NET 4.6?

ReIaxo, 2021-04-30:
We are using .Net 4.6.1

mregen, 2021-05-21:
Hi @Relaxo, how is this possibe? We only support .NET 4.6.2 and later. Also on mono there is no PSS support.

ReIaxo, 2021-06-14:
Sorry for the late reply and for using a not supportet Framework Version.

Until this, it worked for us with Framework Version 4.6.1.
Now we use 4.6.2 and it works just fine. SecurityPolicies.GetDisplayNames returns SecurityPolicies.Aes256_Sha256_RsaPss and the others as expected.

Thank you for your help. I will close this now.

closed 2021-06-14 by ReIaxo

BalajiGuptaK, 2023-09-06:
Hello @mregen,
Is this SecurityPolicies.Aes256_Sha256_RsaPss supported in netstandard2.0?

Because while getting signature value in TryVerifyRSAPssSign method in RsaUtils class throwing an exception "Specified padding mode is not valid for this algorithm." for this.!

Please share your valuable feedback.
Thanks in advance.

mregen, 2023-09-18:
Hi @BalajiGuptaK, the Pss signing support depends on the platform, thats why there is a probe function to check if it is supported. Only on .NET Framework is was well known to work but the probe failed, thats why it is hardwired. If you have time to fix the probe it might be the solution. Please reopen a new issue, as comments in closed issues are normally not noticed.