OPCFoundation/UA-.NETStandard #1373: browser.Browse Program freeze with the the network cable unplugged

Perpete, 2021-04-21:
Hello,

I am currently developing a client OPCUA program for a WAGO PLC.
I am using UA-.NETStandard-Sample as a reference.

While testing my program's operation, I noticed that when I perform a browse operation to retrieve a collection of nodes with the network cable unplugged the program freeze on the 'm_browser.Browse (RootNodeIDParent)' command.
I tried to use a try and catch without success.
There is no error return.
The problem is the same on UA-.NETStandard-Sample.

Is there a solution to detect an error when disconnecting the network cable?

Private Sub FindNodesChildrenTableau (RootNodeIDParent As NodeId)

'Options to be used for the navigation operation
m_browser = New Browser (m_session) With {.BrowseDirection = BrowseDirection.Forward, .ContinueUntilDone = True, .ReferenceTypeId = ReferenceTypeIds.HierarchicalReferences}

'Collection of child variables from parent variable
Dim RdcOrg As ReferenceDescriptionCollection = m_browser.Browse (RootNodeIDParent)
(Freeze program on the line above)

...
...
End sub

I also have an error fetching issue on the await Session.Create () command surrounded by a try and catch.
The error detected for example 'BadCertificateHostNameInvalid' does not go through the Try and Catch.

Try
....
m_session = await Session.Create (m_configuration, endpoint, false, true, Application.ProductName, 60000, identity, nothing)
....
catch
....

Perpete, 2021-04-22:
For await Session.Create () and error handling,
I tested with an old version of Opc.Ua.Client and Opc.Ua.Core (1.4.360.0) dlls. error handling is correct.
The problem is the version 1.4.365.0 dlls that I am using now.

Perpete, 2021-05-09:
Hello,
If the network cable is disconnected just before the 'MySession.ReadValue' command in the monitored items management subroutine, the program freezes.
We have no error feedback.
Error handling works fine for 'MySession.ReadValue' outside of this routine.
In the photo you can see the code in VB with WPF or the program freezes.

[image]

Perpete, 2021-05-12:
**Dim RdcOrg As ReferenceDescriptionCollection = MyBrowser.Browse(RootNodeIDParent)**

Hello,
For the line of code above, the problem does not arise if we wait enough time.
The error occurs in my program after +/- 20 seconds.

For error handling in the monitored items management subroutine, I am wondering if this is not a different thread issue.

AlinMoldovean, 2021-05-25:
Hello @Perpete ,

I made a test using _ConsoleReferenceClient_ project from this repository.
I put a ReadValue call inside `OnMonitoredItemNotification ` and I disconnect the network cable just before the call.
m_session.ReadValue() will trow an exception after some time (~10-20s) just like the call from outside the DataChange notification handler.

[image]

Perpete, 2021-05-26:
Hello,

Thanks for your reply, it was enough to wait as your test shows.

Perpete, 2021-05-31:
Hello,

I did other tests.
Here is what I found.

If you put a breakpoint before a MySession.Write or MySession.Read and you disconnect the cable to the PLC when passing through the breakpoint, the error is detected quickly.

[image]

On the other hand, without a breakpoint with a cable disconnection to the PLC, and a write command, the error trigger is very long (10 minutes in my case).
The program remains frozen for a long time on MySession.Write.
Debug.WriteLine (Now) confirms the blocking time before the error is detected.

[image]

Why this 10-minute time?

Perpete, 2021-07-06:
Hello,
After further testing, I figured out that the 10 minute time was for the 'TransportQuotas.OperationTimeout' property of my client application setup.
This value defaults to 600000.
If I decrease this value, the blocking time of my program with the cable unplugged for a write or read operation of a node also decreases. The blocking time corresponds well to this new value.

I also noticed that the problem occurs if we write or read a node between the disconnection of the cable and the detection by the KeepAlive event. (Except with the breackpoint in the program)

Can you explain to me what determines the detection time of the disconnected cable during a read and write of a node?

Perpete, 2021-07-10:
Hello,
I did other tests with 'UA-.NETStandard-master - Reference Client project' by adding a button to read the nodes on the form.
To perform the test :

- Disconnect the network cable
- Press the read button of a node before triggering the KeepAlive event

In the output window, you can see that an exception is encountered in Opc.Ua.Core.dll.
This exception is generated in a loop.
It can stop after a few seconds or continue until the value of the property 'TransportQuotas.OperationTimeout' (10 minutes).
After all these exceptions, we have the KeepAlive event triggered.
During all this time the program is frozen.

[image]

[image]

Perpete, 2021-07-11:
Hello,
Here are other tests.
I have placed a Debug.Writeline in the following locations:

**Op.Ua.Client.cs**
         public virtual ResponseHeader Read (
             RequestHeader requestHeader,
             double maxAge,
             TimestampsToReturn timestampsToReturn,
             ReadValueIdCollection nodesToRead,
             out DataValueCollection results,
             out DiagnosticInfoCollection diagnosticInfos)
         {
             // 1 *********************************************** ****************************
             System.Diagnostics.Debug.WriteLine ("Invokes the Read service:" + DateTime.Now);
             // ************************************************ *****************************

 **UaSCBinaryTransportChannel.cs**
public IAsyncResult BeginSendRequest(IServiceRequest request, AsyncCallback callback, object callbackData)
        {
            //2 ***************************************************************************
            System.Diagnostics.Debug.WriteLine("Begins send");
            //*****************************************************************************

**UaSCBinaryClientChannel.cs**
 public IAsyncResult BeginSendRequest(IServiceRequest request, int timeout, AsyncCallback callback, object state)
        {
            //3 ***************************************************************************
            //System.Diagnostics.Debug.WriteLine("Sends a request: " + DateTime.Now);
            System.Diagnostics.Debug.WriteLine("Sends");
            //*****************************************************************************

**ChannelAsyncOperation.cs**
public T End(int timeout, bool throwOnError = true)
        {
            //4 ***************************************************************************
            System.Diagnostics.Debug.WriteLine("Response from server: " + DateTime.Now);
            //*****************************************************************************

You can see the course of events in the images below.

Normal procedure:
Detection of a connection error
Reconnection attempt
[image]

Abnormal procedure:
No detection of a connection error
No reconnection attempt
[image]

This problem is random.
Perhaps it is related to the time of the cut?

Perpete, 2021-07-14:
Hello,
Here are further indications from the program with the problematic operation.

In 'private void Session_KeepAlive (Session session, KeepAliveEventArgs e)', this.InvokeRequired always remains true.
This does not allow access to the reconnect request instruction 'm_reconnectHandler.BeginReconnect (m_session, ReconnectPeriod * 1000, Server_ReconnectComplete);'

[image]

Perpete, 2021-07-16:
Hello,
Here is another test with a cable disconnection without reading a node.
We can see in 'private void Session_KeepAlive (Session session, KeepAliveEventArgs e)', that this.InvokeRequired always changes from true to false.
It remains to be determined why this.InvokeRequired in the other tests does not come to false on the new method call after the return.

[image]

AlinMoldovean, 2021-07-16:
Hello @Perpete ,

I tried to investigate the issue, but in my case the delay is only about ~10-20s, 
After a reconnect it goes to normal operation, as it was in the initial test.

Could you please provide a small sample application in C# and the clear steps to reproduce the issue.
I will then further work on this if you can provide this.

Otherwise, it is very difficult to follow a long thread of small code pieces and pictures...

Regards,
Alin

Perpete, 2021-07-16:
Hello,

To perform the test, I used the UA-.NETStandard-master solution with the Reference Client project 'starter project.
I added a button to read the TreeView selection node.
(See the animated image).
The code linked to the button (private void btRead_Click (object sender, EventArgs e) can be retrieved from a previous comment.
I validated the display of the trace '<TraceMasks> 519 </TraceMasks>' in 'Quickstarts.ReferenceClient.Config'.

**To create the problem:**
(Freeze main program 10 minutes)

1. Show output window for debugging and follow trace display
2. Disconnect the network cable to the PLC
3. Wait for the line 'Channel 0 in Connecting state' to appear in the output window.
4. Press the button to read the node

[image]

**So as not to have any problem:**
(Freeze main program 10 to 20 seconds)

1. Show output window for debugging and follow trace display
2. Disconnect the network cable to the PLC
3. Quickly press the play button on the node before the "Channel 0 in Connecting state" line appears in the output window.

[image]

I hope I have been clear enough so that you can reproduce the problem.
At home, with this procedure, I reproduce the problem every time.

Thanks for your help.

AlinMoldovean, 2021-07-19:
Hello @Perpete ,

I tried again to reproduce the issue using the provided steps.
I used ReferenceClient on the testing machine and ReferenceServer on a network PC.
I clicked the "_Read_" button right after the "_Channel 0 in Connecting state_" log message.
The result was that the button-Read operation was failing immediately with a BadConnectionClosed error.

I was not able to use a WAGO PLC server for testing, so I used ReferenceServer of this repo.
I do not know if that is the reason why the condition does not replicate.

[image]

Perpete, 2021-07-19:
Hello,

Thanks for your testing.

For my tests, I use a 750-8100 PLC from the company Wago.
For the network link between the PC and the PLC, I use an EMINENT EM1017 R2 USB 3.1 network adapter with the correct drivers.

I notice on your image that you do not have the same sequencing for the creation of the problem as me.
The line 'Channel 0 in connecting state' comes too late compared to my sequencing.
In your test, it comes after the reconnection test phase.
In this case, you will not have the problem.
The 'Channel 0 in connecting state' line must arrive before the reconnection attempt to produce the problem.
In this case, the problem recurs every time.

[image]

Perpete, 2021-07-29:
Hello,

Here is another display of traces which shows the looping on the secondary thread no longer giving access to the main thread.
The main program is frozen.

[image]

Perpete, 2021-08-04:
Hello,

To help you, here is a screen corresponding to the parallel stacks when the main program is frozen during the looped passage in 'private void Session_KeepAlive (Session session, KeepAliveEventArgs e)'.
(Reading a node with the cable disconnected following illustrated sequence).

[image]

Perpete, 2021-08-07:
Hello,
Here is a comparison of the sequence of events displayed in the visual studio output window.
Compared to the existing schedule, I have added other messages to follow the process of unfolding events.

[image]

Perpete, 2021-08-15:
Hello,
I did other tests by adding a write node button on the form.
I have exactly the same problem.

[image]

Shouldn't we simply prevent the action of the flow of the read and write service invocation functions     to Opc.UA.Client.cs class when the cable is unplugged?

Perpete, 2021-08-24:
Hello,
The solution proposed by mrsuciu is to call the reads / writes from a different thread than the main thread, or to use their asynchronous versions (See issue # 1485)
I have tried this solution.
It solves the problem.

For information, here is the asynchronous reading code used for the test.

`

        private void btRead_Click(object sender, EventArgs e)
        {

            //Lecture du node sélectionné sur le treeview

            ReferenceDescription noderead = BrowseCTRL.BrowseCTRL.SelectedNode;

            // build list of attributes to read.
            ReadValueIdCollection valueIdsToRead = new ReadValueIdCollection();

            if (m_NodeSelectId is null && noderead is null)
            {
                return;
            }

            //**********************************
            System.Diagnostics.Debug.WriteLine("Request to read a node " + DateTime.Now);
            //**********************************

            if (m_NodeSelectId is null)
            {
                m_NodeSelectId = (NodeId)noderead.NodeId;
            }

            ReadValueId attributeToRead = new ReadValueId();
            attributeToRead.NodeId = m_NodeSelectId;
            attributeToRead.AttributeId = Attributes.Value;
            valueIdsToRead.Add(attributeToRead);

            m_session.BeginRead(null, 0, TimestampsToReturn.Both, valueIdsToRead,OnReadComplete, m_NodeSelectId.Identifier);
           
        }

        private void OnReadComplete(IAsyncResult result)
        {
            // Need to make sure this method is called on the UI thread because it updates UI controls.
            if (this.InvokeRequired)
            {
                // Asynchronous execution of the AsyncCallback delegate.
                this.BeginInvoke(new AsyncCallback(OnReadComplete), result);
                return;
            }

            try
            {
                DataValueCollection results;
                DiagnosticInfoCollection diagnosticInfos;
                ResponseHeader responseHeader = m_session.EndRead(result, out results,out diagnosticInfos);

                if (StatusCode.IsGood(results[0].StatusCode) && StatusCode.IsGood(responseHeader.ServiceResult))
                {
                    lbRead.Text = Convert.ToString(results[0].Value);
                }
                else
                {
                    MessageBox.Show("Problème de lecture avec la variable", "Erreur lecture", MessageBoxButtons.OK, MessageBoxIcon.Error);
                }
            }
            catch (Exception ex)
            {
                MessageBox.Show("Erreur lecture. Error: " + ex.Message);
            }
        }

`

closed 2021-08-24 by Perpete