OPCFoundation/UA-.NETStandard #1377: License file should include redist lic for NuGet packages

barnstee, 2021-04-23:
**Type of Issue**
- [ ] Bug 
- [ ] Enhancement 
- [X] Compliance 
- [ ] Question 
- [ ] Help wanted

**Describe the Issue**
Right now, only the source lic is described in the lic file, but folks continue to be confused about the lic for the NuGets, which is covered by the rest lic. Please include it here on GitHub.

**To Reproduce**
N/A

**Expected behavior**
All licenses are described.

**Screenshots**
N/A
**Log output**
N/A
**Code snippets**
N/A
**Environment (please complete the following information):**
N/A
**Additional context**
N/A

mregen, 2021-04-29:
The description needs an update: https://github.com/OPCFoundation/UA-.NETStandard/blob/gh-pages/index.htm
Second option is to append the redist license to the license.txt to clarify under which license nuget packages are distributed.

mregen, 2022-04-01:
see #1763 and #1764

jh-isw, 2022-09-26:
That the Redist License is removed means, when I build a Client application, I have to disclose the source code?

[image]

When I add the OPCFoundation.NetStandard.Opc.Ua.Client NuGet-Package to any project, I have to accept all the licenses of the packages shown in the screenshot.

OPCFoundation.NetStandard.Opc.Ua.Core has the Dual-License attached and that would mean GPL for me as a non-member (although it only talks about sources, not compiled libs).

Was that the intention, that only corporate members can build applications without having to show their code??

mregen, 2022-09-27:
Hi @jh-isw, there is no change in licensing, it just wasn't obvious when reading the redist license that it required to look at the source code license to determine the license for a package. Now the packages point directly to the license that is applicable. 
So yes, the dual license is and was always applicable for the core package. For what this means exactly for you as non corporate member please get legal advice. 
The OPC Foundation is dedicated to support the open source community including non members by means of the GPL 2.0 license for the source code and the Nuget packages.
Corporate members have the option to pick the RCL license for their open source projects or applications.

jh-isw, 2022-09-30:
Hello @mregen , thank you for clarifying!

mregen, 2022-12-11:
All license files are up to date.

closed 2022-12-11 by mregen