OPCFoundation/UA-.NETStandard #1378: SecurityConfiguration.RejectedCertificateStore.StoreType and StorePath : Error NullReferenceException

Perpete, 2021-04-24:
Hello,

am using the Opc.Ua.client and Opc.Ua.Core 1.4.365.23 dlls.
I also did some tests with version 1.4.360.0.
In both cases, I encountered the same problem.
When setting the StoreType and StorePath properties of RejectedCertificateStore outside of the ApplicationConfiguration () constructor, we get the error:
NullReferenceException: The object reference is not set to an instance of an object.
This error is not present for the same definitions of the other certificate folders (ApplicationCertificate, TrustedIssuerCertificates ..).

[image]

To avoid this problem, I had to declare the properties of RejectedCertificateStore in the ApplicationConfiguration () constructor.

[image]

Is this normal?

Perpete, 2021-04-25:
Corrected with the latest dlls updates.

mregen, 2021-04-29:
@Perpete the rejected store is optional, thats why it may behave different than other stores. Doing the config manually has some pitfalls to overcome, please use the supported configuration API.

Perpete, 2021-04-30:
I installed the nuguet package and it works.
Thank you.

closed 2021-04-30 by Perpete