OPCFoundation/UA-.NETStandard #1379: Certificate not found error with CertificateFactory.CreateCertificate (Obsolete)

Perpete, 2021-04-25:
Hello,

I am using the latest Opc.Ua.Client and Opc.Ua.core Dlls.
I was creating a certificate using BouncyCastle.Crypto.dll and the CertificateFactory.CreateCertificate method.
Now this method is obsolete.
If I continue to use it, I get an error message when creating my session. (Certificate is not trusted).
However, the certificate is indeed found after its creation.
This problem did not exist with the old dlls.

I think I will have to use the new CertificateBuilder method to create the certificate.
I tried to find examples of using this new method on the net. Without success.
To solve my problem, can you help me understand this new method, if possible with examples?
Thank you.

[image]

[image]

mregen, 2021-04-29:
@Perpete , there is maybe an issue with the subject, the obsolete function maps to the new API, so it should work. For loading of certs I recommend to use the application config API, there are some ways private keys not properly loaded or certs not found implementing it your self.

Perpete, 2021-04-30:
I installed the nuguet package and used the configuration api
I am using a configuration.xml file to create my certificate.
It works very well.
Thank you.

closed 2021-04-30 by Perpete