OPCFoundation/UA-.NETStandard #1382: Application LoadApplicationConfiguration Wait method not found (System.Threading.Tasks.Task)

Perpete, 2021-04-28:
Hello,

In my program, I load a configuration file with the method:
application.LoadApplicationConfiguration (false) .Wait () ;.
The following error is raised:
Method not found: 'System.Threading.Tasks.Task`1 Opc.Ua.ApplicationConfiguration.Load (System.IO.FileInfo, Opc.Ua.ApplicationType, System.Type, Boolean)'.
If I delete the Wait, I no longer have an error message but the configuration does not load.
The configuration file is a file used in UA-.NETStandard-Samples-master and placed in my project.

The CheckApplicationInstanceCertificate method with the Wait causes the same error.

I am using your latest dlls with the 4.8 framework and visual studio 16.9.4.
Am I missing a reference?
I enclose my test program for possible tests.

[image]

[Essai_OPC_Config.zip](https://github.com/OPCFoundation/UA-.NETStandard/files/6390753/Essai_OPC_Config.zip)

mregen, 2021-04-29:
@Perpete , I suspect you somehow added teh Opc.Ua.Core assemblies directly to your project. This means you don't get all dependencies. Please use the supported Nuget packages in your project.

Perpete, 2021-04-30:
I installed the nuguet package and it works.
Thank you.

closed 2021-04-30 by Perpete