OPCFoundation/UA-.NETStandard #1390: Getting "The empty string '' is not a valid local name" while Nodestate.SaveAsXml

atodkar, 2021-04-29:
Hello,

After loading NodeState into AddressSpace, I am trying to serialize individual NodeState into XML and facing below Exception.

`
System.ArgumentException: The empty string '' is not a valid local name.
   at System.Xml.XmlWellFormedWriter.WriteStartElement(String prefix, String localName, String ns)
   at System.Xml.XmlWriter.WriteStartElement(String localName, String ns)
   at Opc.Ua.XmlEncoder.Initialize(String fieldName, String namespaceUri) in C:\opcua\UA-.NETStandard\Stack\Opc.Ua.Core\Types\Encoders\XmlEncoder.cs:line 111
   at Opc.Ua.XmlEncoder..ctor(XmlQualifiedName root, XmlWriter writer, ServiceMessageContext context) in C:\opcua\UA-.NETStandard\Stack\Opc.Ua.Core\Types\Encoders\XmlEncoder.cs:line 75
   at Opc.Ua.NodeState.SaveAsXml(ISystemContext context, Stream ostrm) in C:\opcua\UA-.NETStandard\Stack\Opc.Ua.Core\Stack\State\NodeState.cs:line 549
   at PDM.AggregatingServer.OPC.OnboardingNodeManager.OnboardConnection(IDictionary`2 externalReferences, DirectoryInfo nodeSetFolder)
`

Tried saving base types as well but getting same issue. Some of nodes from BaseTypes get save though.

AlinMoldovean, 2021-05-06:
Hi @atodkar,

The problem seems to be that `SymbolicName `property of the node is not set, i.e. is empty string.

atodkar, 2021-05-06:
Thanks @AlinMoldovean, looks like that would be the reason. But SymbolicName is an optional attribute and I see many nodes from OPCFoundation Companion specification XMLs are missing it. 

Do you think this should be handled in SaveAsXml function?

AlinMoldovean, 2021-05-07:
Hi @atodkar ,

A better option is to use the `NodeStateCollection.SaveAsNodeSet2(ISystemContext context, Stream ostrm)` method.
It also allows to save a set of nodes instead of a single NodeState.

AlinMoldovean, 2021-05-07:
Hi @atodkar ,

A better option is to use the NodeStateCollection.SaveAsNodeSet2(ISystemContext context, Stream ostrm) method.
It also allows to save a set of nodes instead of a single NodeState.

atodkar, 2021-05-07:
Thanks.

Yes, but this is also serializing alias and other details which is making XML bulky. I was looking for lighter option. I also tried SaveAsBinary but that one I am not able to reconstruct correctly. Do you know what would be the correct way to SaveAsBinary and then LoadNode using Binary decoder to reconstruct that Node?

AlinMoldovean, 2021-05-14:
@atodkar ,

SaveAsBinary produces a output that is difficult to interpret and reconstruct.

closed 2021-05-24 by AlinMoldovean