OPCFoundation/UA-.NETStandard #1404: BadIdentityTokenInvalid only when attempting to connect with User Name & Password

dekenless, 2021-05-17:
**Type of Issue**
- [ ] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [X ] Question 
- [ ] Help wanted

Hi,

I am connecting to an OPC Server which all works fine when anonymous connections are permitted. However, when I try using a user name and password I receive the following:

`Opc.Ua.ServiceResultException: BadIdentityTokenInvalid
   at Opc.Ua.ClientBase.ValidateResponse(ResponseHeader header)
   at Opc.Ua.SessionClient.ActivateSession(RequestHeader requestHeader, SignatureData clientSignature, SignedSoftwareCertificateCollection clientSoftwareCertificates, StringCollection localeIds, ExtensionObject userIdentityToken, SignatureData userTokenSignature, Byte[]& serverNonce, StatusCodeCollection& results, DiagnosticInfoCollection& diagnosticInfos)   at Opc.Ua.Client.Session.Open(String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales, Boolean checkDomain)    at Opc.Ua.Client.Session.Create(ApplicationConfiguration configuration, ITransportWaitingConnection connection, ConfiguredEndpoint endpoint, Boolean updateBeforeConnect, Boolean checkDomain, String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales)
 `

The code I am using to connect with a password is:

`  selectedEndpoint.UserIdentityTokens.Add(new UserTokenPolicy(UserTokenType.UserName) { });
  user = new UserIdentity(UserName, Password);
  await Session.Create(Config, configuredEndpoint, false, true, SessionName, 60000, user, null);`

Any thoughts?

Environment (please complete the following information):
 - OS:  Windows 10
 - Development environment: Visual Studio 2019 16.8.3
 - Runtime:  .NET Core 3.1
 - OPC Package Release Version 1.4.362.42
 - Server: Kepware

AlinMoldovean, 2021-05-19:
@dekenless ,

You need to send the user and password in  at `Session.Create()` without adding it in _selectedEndpoint.UserIdentityTokens_
If it is not yet there, it means the server does not support that type of authentication.

```
                    // Create the session
                    Session session = await Session.Create(
                        m_configuration,
                        endpoint,
                        false,
                        false,
                        m_configuration.ApplicationName,
                        30 * 60 * 1000,
                        new UserIdentity("userName", "password"),
                        null
                    );

```

dekenless, 2021-05-19:
Thanks! That addressed the issue.

closed 2021-05-19 by dekenless