OPCFoundation/UA-.NETStandard #1416: getting event in subscription when there is no change in data from OPC UA server

rathinams, 2021-05-31:
getting event in subscription when there is no change in data from OPC UA server - [ ] Help wanted

I have a .net core MVC application which is communicating to the PLC.
1. I create a OPC UA session
2. I start creating monitored items for subscription with a publishing interval of 1000.
3. then i create a subscription with all the monitored items
4.  immediately I am getting events returned without change in data of nodes from the server.

private List<MonitoredItem> monitoredItems = = new List<MonitoredItem>();

 private async Task<Session> GetSessionAsync(bool IsPLCServer)
{
                                session = await Session.Create(_appConfiguration,
                                                     endpoint,
                                                     false,
                                                     _appConfiguration.ApplicationName + "_session",
                                                     (uint)_appConfiguration.ClientConfiguration.DefaultSessionTimeout,
                                                     new UserIdentity(new AnonymousIdentityToken()),
                                                     null);
                    AddSubscription();
                    CreateSubscription(session, 1000, true);
}

private void CreateSubscription(Session session, int publishingInterval, bool publishEnabled)
        {
 var  subscription = new Subscription(session.DefaultSubscription)
                {
                    PublishingInterval = publishingInterval,
                    PublishingEnabled = publishEnabled
                };
                monitoredItems.ForEach(i => i.Notification += OnNotification);
                subscription.AddItems(monitoredItems);
                session.AddSubscription(subscription);
                subscription.Create();
            }

private void AddMonitoredNodes(dynamic Nodes)
        {

                MonitoredItem monItem = new MonitoredItem();
                foreach (var node in Nodes)
                {
                    monItem.DisplayName = node.Key;
                    monItem.StartNodeId = node.Value;
                    monitoredItems.Add(monItem);
                }

            }

AlinMoldovean, 2021-05-31:
Hello @rathinams ,

I do not understand what is the actual question.

When creating a monitored item for a node you get the initial data value as a DataChange notification.
After that, the server will send notifications only when the value of the monitored node has changed.

rathinams, 2021-06-01:
Hi @AlinMoldovean 

Yes thats right am getting a data change notification initially when i create a monitored item, how to avoid this initial data change notification?

AlinMoldovean, 2021-06-02:
Hi @rathinams ,

When an OPC UA client creates a MonitoredItem, the UA Server will send a notification with the initial value within the first publishing interval. 
That is the normal behavior requested by the OPC UA protocol specification.

You cannot change this behavior in the Server component of this repository without modifying the internal implementation.
That will make the server uncompliant

closed 2021-06-02 by AlinMoldovean

rathinams, 2021-06-02:
@AlinMoldovean  Thanks for the information.
When a monitored item is created three thing happens right (correct me here please)
1. initial value notification with the publishing interval
2. continued value notification for every publishing interval
3. when there is a change in value then notification

AlinMoldovean, 2021-06-02:
@rathinams ,

It is only:
  1. - initial value notification within the first publishing interval
  2.  - notification when there is a change in value

rathinams, 2021-06-02:
@AlinMoldovean 
Thanks where can i learn more about that please suggest.

AlinMoldovean, 2021-06-02:
@rathinams , you can check the specs here:
https://reference.opcfoundation.org/Core/docs/Part4/5.12.1/