OPCFoundation/UA-.NETStandard #1418: Can an OPC UA client establish a reaction based update rate with an OPC UA server?

JustinLipps, 2021-06-04:
Type of Issue
[ ] Bug [ ] Enhancement [ ] Compliance [X] Question [ ] Help wanted

Is it possible to configure or implement an OPC UA client in such a way that it responds to data changes within the server (where the server would actually send a value back (create another network packet) when the data has changed) or is the only possibility a polling based messaging system defined by the "Publishing Interval" of each tag?

Thanks!

AlinMoldovean, 2021-06-08:
Hello @JustinLipps ,

The standard mechanism in OPC UA protocol for the efficient value change notifications is the Subscription\MonitoredItems model.
Another possible option is a cyclic Read, but it should be avoided.

JustinLipps, 2021-06-08:
Ok, perfect, thanks for taking the time to answer that!

closed 2021-06-09 by AlinMoldovean