OPCFoundation/UA-.NETStandard #1419: Request for enhancement: Add Support for custom certificate stores

mheege-abb, 2021-06-06:
The current SDK supports only Windows and Directory certificate stores. For security reasons, it can be necessary to use custom certificate stores. I have a local implementation that extends this support. If you add me to the contributors, I can create a pull request.

mheege-abb, 2021-06-08:
I guess my suggestion would solve https://github.com/OPCFoundation/UA-.NETStandard/issues/1202, too. In this issue @barnstee implemented a suggestion for TPM key, but the link does not work. It would we good if he could share his code if it is still available.

barnstee, 2021-06-08:
The code is now hopelessly out of date but a custom cert library is on the backlog. @mregen?

closed 2022-01-17 by mregen