OPCFoundation/UA-.NETStandard #1425: Why is Byte[] returned as ByteString on a read?

hermoter, 2021-06-15:
**Type of Issue**
- [ ] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [ X ] Question 
- [ ] Help wanted

**Describe the Issue**
When I do a read on a node with a data type of Byte and rank == 1 (one-dimensional array), I get a ByteString back.
This is unexpected.

When I test this with a OPC UA client, the value is displayed as a ByteString, even though the node attributes clearly show the datatype being Byte and the rank == 1.

**Expected behavior**
I would expect a true array of Byte.
I tested this with another OPC UA server (https://open62541.org/) which returns the value as a true Byte[] and not a ByteString.

**Screenshots**
This shows the incorrect datatype ByteString being returned for a node with a Byte[] definition:
[image]

This shows the correct datatype Byte[] being returned from another open source OPC UA Server:
[image]

Could you please explain this behavior?
Is this intentional or a bug?

AlinMoldovean, 2021-06-15:
Hello @hermoter ,

The _ByteString_ and _Byte[]_ types in OPC UA are identical and interchangeable.
This .NET stack handles Byte[] values as ByteString. This is acceptable and escribed in the specification:

https://reference.opcfoundation.org/Core/docs/Part4/5.10.4/

[image]

hermoter, 2021-06-15:
Thanks for the quick response! Answer accepted.

closed 2021-06-15 by hermoter