OPCFoundation/UA-.NETStandard #1426: UA .NETStandard on GitHub does not have Publisher and Subscriber Sample Projects

aehomay, 2021-06-15:
**Type of Issue**
- [ ] Bug 
- [X ] Enhancement 
- [ ] Compliance 
- [ ] Question 
- [ ] Help wanted

**Describe the Issue**

The source code for UA .NETStandard on the OPC Foundation website ([here](https://opcfoundation.org/developer-tools/samples-and-tools-unified-architecture/net-stack-and-sample-applications/)) includes two sample projects ConsoleReferencePublisher and ConsoleReferenceSubscriber to illustrate how to bring up an OPC UA Publisher and Subscriber. However, these projects are missing on the GitHub repository.

AlinMoldovean, 2021-06-16:
Hello @aehomay ,

ConsoleReferencePublisher\Subscriber projects are included in this repository in "_UA PubSub Preview.sln_" and they are located here:

https://github.com/OPCFoundation/UA-.NETStandard/tree/master/Applications/ConsoleReferencePublisher

https://github.com/OPCFoundation/UA-.NETStandard/tree/master/Applications/ConsoleReferenceSubscriber

closed 2021-07-13 by AlinMoldovean

aehomay, 2021-07-14:
The question is not where the publisher and subscriber sample applications are located. I didn`t ask this I reported an issue on Publisher and Subscriber please read it before closing it.