OPCFoundation/UA-.NETStandard #1429: Euromap 77 question

matteomessmer, 2021-06-18:
**Type of Issue**
- [ ] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [x] Question 
- [ ] Help wanted

**Describe the Issue**

Hi, I cannot find any technical documentation on how to interface with a machine that uses Euromap 77. The machine's producer told me that nodes IDs can change over time so I have to adapt the client I developed. Does someone have docs, a simulator or any other useful information about Euromap 77 that developers can use? I have the docs but they are not technical, just overall introduction and something about the machine, but that doesn't help much the development.

What I'm trying to understand is how I can use this library for reading nodes from an Euromap 77 machine.

MD-V, 2021-06-19:
Hello @matteomessmer,

as there is no ModelDesign you can feed to the ModelCompiler you will have to do some manual work.

You have multiple options:

You can try to run the https://github.com/OPCFoundation/UA-.NETStandard-Samples/tree/master/Samples/NetCoreComplexClient on the server and see how it can handle the Euromap 77 data types.

Alternatively you can implement all custom types you need to handle as an `IEncodable`.
Examples:
[https://github.com/OPCFoundation/UA-.NETStandard/issues/793#issuecomment-531771132](1) [https://github.com/OPCFoundation/UA-.NETStandard/issues/658#issuecomment-465665115](1)

eoursel, 2021-06-19:
@matteomessmer some of my colleagues in the US have written a euromap 77 simulator 2 years ago in nodered. I will check the status with them and will come back with an answer in a few days. I must check internally if I can disclose the source code. 

Regards

matteomessmer, 2021-06-22:
@MD-V Thank you, I'll take a look at those 
@eoursel Thank you very much, it would be a really good resource for the community

matteomessmer, 2021-08-04:
I have some updates about this and a question. 
In UA Expert I can see all the nodes (IDs, values etc.) but from the machine documentation I've learned that I cannot trust the nodes ID to be immutable over time. I have to create a "protocol" on the machine (which I managed to do) and get the nodes values through that.
In UA Expert I can see this "protocol" by dragging and dropping the machine object in the event view tab. The "protocol" is inside this object in the event view tab and inside it there are all the nodes I need. 
Is there a way to get the object I'm seeing in the event view tab using this library?

MD-V, 2021-08-04:
Hello @matteomessmer,

unfortunately I don't understand your question. Can you clarify "protocol" with some screenshots from UAExpert? You can directly paste images in here.

matteomessmer, 2021-08-04:
Hi @MD-V, 
on the machine I'm using you have to create what is called "protocol" from the panel of the machine, then within this "protocol" I'm going to insert the nodes that interest me. I do not know if I can share the documentation because I can not find it online. 

However to give an idea this is a screenshot of ua expert with the simulator server, it is not the same but the object highlighted is similar to the real situation (now I can not take a screenshot of that because I do not have access but as soon as I can I'll add that). 

[image]

In the example screenshot inside "TransitionEventType" there are Transition, FromState and ToState, while in the real situation I see all the nodes I need to read, i.e. those inserted in the "protocol"

MD-V, 2021-08-05:
@matteomessmer,

What you see inside the eventview is (what I beliebe) custom browse logic of the UAExpert:
[image]

To get those informations you will have to manually browse (ReferenceTypes: HasProperty, HasSubType) the BaseEventType (see NodeId in picture) with the Browser class of the libary.
[image]

matteomessmer, 2021-08-05:
Thank you @MD-V, I'll try the Browser class

matteomessmer, 2021-08-05:
@MD-V Thank you, I think this is the right way, as soon as I try it on with the customer machine, I'll write here the result 👍🏼

benjamin1988, 2021-08-17:
Hi @matteomessmer,

we are also writing em77 drivers. And as there could be dynamic nodes the node id's cannot be trusted. So we work with the browse names and the parents Nodes. 
For example if you want to read the current JobName you first have to browse the node with the BrowseName ActiveJob in the EM83 namespace and afterwards the underlying child node JobName. 

To read the event information of your TransitionEvenType you can browse the underlying child nodes Transition,... and add them to the select clause with their NodeId out of the event definition.
[image]

May I could help you a little bit.

matteomessmer, 2021-08-23:
Hi @benjamin1988, yes that's exactly what I did. I think this is the right approach and it also works on the machine of my customer. Also thanks again @MD-V 

@benjamin1988 While I'm at it, I'd like to ask you if you know how alarms can be read. I haven't had a chance to see with the customer's machine yet as it is not yet in use, but on UAExpert I have seen that you can see alarms in events. What I would like to implement in my program, in addition to reading the nodes I am interested in, is to know whether any alarms have been triggered.

MD-V, 2021-08-23:
@matteomessmer 
To receive Alarms and Events you will have to create a subscription and one or more MonitoredItems with attribute id EventNotifier (see [here](https://github.com/grafana/opcua-datasource/blob/b3a78e8da6f699f9a272da459771d54f38cf9531/backend/EventSubscription.cs#L156))

Also you will need to specify which attributes you want to receive in the event filter.

benjamin1988, 2021-09-08:
@matteomessmer 
As MD-V mentioned, just use an event notification with the alarm type. Until now we don't have any machine which raises the alarm as OPC-UA alarm. They just raise an event of type AlarmConditionType. From my point of view the alarms should be shown in Alarm view of OPC-UA Expert.

OleksandrPerederiy, 2021-12-16:
Hello @matteomessmer,
I am also working on the development of the OPC UA Service for connecting machines with Euromap 77. As you say, for development purposes it would make sense to use a machine simulator with Euromap 77. Can you recommend such an OPC UA Server Euromap 77 simulator to me?

benjamin1988, 2021-12-16:
Hello @OleksandrPerederiy,

i don't think that a free euromap 77 simulator is available. Why do you want to implement a EM77 client? May you have contact to a manufacturer of an injection molding machine?

With kind Regards,
Benjamin

matteomessmer, 2021-12-16:
Hi @OleksandrPerederiy, I didn't find an Euromap 77 simulator, instead I used an OPC UA Server Simulator to test the browse of the nodes. If you can find a Euromap 77 simulator let me know :)

OleksandrPerederiy, 2021-12-17:
Hi @benjamin1988 and @matteomessmer 
thanks you for quick replies!
We already operate our own OPC UA client and only need to expand it for Euromap 77.
Yes, I will contact a manufacturer of an injection molding machine.

Best regards,
Alex

OleksandrPerederiy, 2021-12-17:
If you have a Siemens PLC you can try it.
But it doesn't work with every PLC, it has to be a slightly stronger model.
 [https://support.industry.siemens.com/cs/document/109755133/siemens-opc-ua-modeling-editor-(siome)-for-implementing-opc-ua-companion-specifications?dti=0&lc=en-WW](url)

closed 2023-06-21 by EthanChangAED