OPCFoundation/UA-.NETStandard #1432: Client Accept Expired Certificate Help

A-Kvisgaard, 2021-06-25:
**Type of Issue**
- [ ] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [X] Question 
- [X] Help wanted

**Describe the Issue**
I'm new to OPC-UA and this Libary, so sorry if this should be self-explanatory. 
I'm trying to connect to a server that has an expired certificate with a Session. With my current setup, I get BadCertificateTimeInvalid error which let me to issue #1152. Here It's explained that "CertificateValidator can suppress expired cert errors: ((id.ValidationOptions & CertificateValidationOptions.SuppressCertificateExpired)". But I can't for the life of me figure out where to set this option. Any help would be greatly appreciated 

**Code snippets**
Client
```C#
public OpcClient(string endpoint, 
            int maxRetries = DefaultVariables.DefaultMaxConnectionRetries)
        {
            // operating is set to true, to indicate that the client is successfully running
            Operating = true;

            // current retries is initialized to 0
            CurrentRetries = 0;

            // endpoint is assigned to URL
            URL = endpoint;

            // max retries 
            MaxRetries = maxRetries; 

            // initialize empty dictionary
            References = new Dictionary<ReferenceDescription, MonitoredItemNotificationEventHandler>();

            // assign the default callbacks
            DefaultKeepAliveCallback = new KeepAliveEventHandler(DefaultKeepAliveCallbackMethod);
            DefaultNotificationCallback = new MonitoredItemNotificationEventHandler(DefaultCallbackMethod);

            // create the application client type instance with certificate
            ClientConfiguration = ConfigUtils.CreateAppConfig(ApplicationType.Client);
            ConfigUtils.BuildAppInstance();

            Console.WriteLine( "Initialized client on endpoint: " + URL);
        }

 public bool Connect(KeepAliveEventHandler callback = null, uint timeout = DefaultVariables.DefaultSessionTimeout, int keepAliveInterval = DefaultVariables.DefaultKeepAliveInterval)  
        { 
            try  
            {

                // initializes the endpoint by verifying that the destination exists
                // this will fail if the URL is wrong or the server is down
                EndpointDescription endpoint = CoreClientUtils.SelectEndpoint(URL, useSecurity: true, DefaultVariables.DefaultOperationTimeout);

                // creates the client session with the timeout provided. The session is asynchronous and runs in the background.
                ClientSession = Session.Create(ClientConfiguration, new ConfiguredEndpoint(null, endpoint, EndpointConfiguration.Create(ClientConfiguration)), false, "", timeout, null, null).GetAwaiter().GetResult();

                // if no callback is assigned, use the default keep alive event handler
                if (callback == null) callback = new KeepAliveEventHandler(DefaultKeepAliveCallbackMethod);

                // else, assign the default keep alive handler to the provided callback
                else DefaultKeepAliveCallback = callback;

                // add the callback to the client session
                ClientSession.KeepAlive += callback;    

                // start the session with the provided keep alive interval
                ClientSession.KeepAliveInterval = keepAliveInterval;
                
                // reset current retries to 0
                CurrentRetries = 0;

[... 157 lines cut ...]
            // assign the configuration to CurrentConfiguration
            CurrentConfiguration = config;

            // assign the type to CurrentType
            CurrentType = type;

            // assign the name to CurrentAppName
            CurrentAppName = applicationName;

            config.SecurityConfiguration.ApplicationCertificate.ValidationOptions = CertificateValidationOptions.SuppressCertificateExpired;

            // return ApplicationConfiguration
            return config;
        }

```

**Environment (please complete the following information):**
 - OS: Windows 10 
 - Development environment: Visual Studio 2019 16.10.2
 - Runtime: net5.0
 - OPC Package Release Version 1.4.366.38
 - Component: Opc.Ua.Client
 - Server: Server with expired Ceret
 - Client: Defiend client above

MD-V, 2021-06-25:
Hello @A-Kvisgaard,

you need to add custom Certificate Validation on your client.

Check https://github.com/OPCFoundation/UA-.NETStandard/blob/e34eb9b512723e474320300596e92cdb2579f0dc/Applications/ConsoleReferenceClient/UAClient.cs#L53 and https://github.com/OPCFoundation/UA-.NETStandard/blob/e34eb9b512723e474320300596e92cdb2579f0dc/Applications/ConsoleReferenceClient/UAClient.cs#L464 for an example.

As I can see you have implemented this already. Try to make the following code change:

 config.CertificateValidator.CertificateValidation += (s, e) => {  
                    e.Accept = true; 
                };

A-Kvisgaard, 2021-06-25:
Thank you @MD-V I can connect now, cheers!

closed 2021-06-25 by A-Kvisgaard