OPCFoundation/UA-.NETStandard #1435: OPC Server as a windows service can't get correct 'xxx.Config.xml'

wwwolfsasa, 2021-06-27:
**Type of Issue**
- [x] Question 

**Describe the Issue**
I build a opc server as a windows service.
When it tried to get 'xxx.Config.xml', then it got a error log about "Load Application Configuration: Could not find file 'C:\WINDOWS\system32\xxx.Config.xml'".

After that I found the reason how it happened, following code is the reason:
This code is at class 'Opc.Ua.Utils' Line 714.
```c
localFile = new FileInfo(Utils.Format("{0}{1}{2}", Directory.GetCurrentDirectory(), Path.DirectorySeparatorChar, filePath));
```
'Directory.GetCurrentDirectory()' causes this problem.

Is there possible change to use Assemble to get the exactly where it is ?? like following code:
```c
localFile = new FileInfo(Utils.Format("{0}{1}{2}", new System.IO.FileInfo(typeof(Utils).Assembly.Location).DirectoryName, Path.DirectorySeparatorChar, filePath));
```

thank you.

**To Reproduce**
Steps to reproduce the behavior:
1. Get 'xxx.Config.xml' Before Starting OPC Server

**Environment (please complete the following information):**
 - Component: [Opc.Ua.Core]

AlinMoldovean, 2021-06-30:
Hello @wwwolfsasa ,

Are you using .NET Framework or .NET Core runtime?
For .NET Framework there is a logic to further check `Path.GetDirectoryName(Assembly.GetExecutingAssembly().Location)` that should cover your case.
https://github.com/OPCFoundation/UA-.NETStandard/blob/d972f80c2facb2de2a7471264537616af3caf394/Stack/Opc.Ua.Core/Types/Utils/Utils.cs#L715-L726

If you are using .NET Core, could you check if removing the #if NETFRAMEWORK condition solves the issue?

wwwolfsasa, 2021-07-04:
Hi @AlinMoldovean ,

Yes, I am using .NET Core.
Thank you for your response, I will try your away in my code.

mrsuciu, 2021-09-15:
Please re-open if this is still an issue

closed 2021-09-15 by mrsuciu