OPCFoundation/UA-.NETStandard #1437: Keep the connection open or reconnect each time if the reading frequency is low?

ShuhuaGao, 2021-06-29:
**Type of Issue** 
- [x] Question 
- [x] Help wanted

**Describe the Issue**
This is not a technical issue. Just a question, and hope you can give some suggestions.

Supposing the OPCUA client needs to read some data from the server every 2 minutes, shall I keep the connection open for the whole life of the application, or shall I close the connection once reading is finished, and then reconnect it every 2 minutes? What is the best practice? 

Thanks for any advice.

AlinMoldovean, 2021-06-30:
Hello @ShuhuaGao ,

Four such seldom operations is better to create and close a Session instead of blocking allocated resources on network and server side.

ShuhuaGao, 2021-06-30:
@AlinMoldovean . Thanks for the reply. I have the same feeling. In short, in such low-frequency communication, it is recommended to start/close a session when needed rather than keep a long-running session, right?

AlinMoldovean, 2021-06-30:
@ShuhuaGao  yes, that is correct.

closed 2021-06-30 by ShuhuaGao