OPCFoundation/UA-.NETStandard #1440: node value updated twice in the same sampling interval

Praveen-g99, 2021-07-01:
SDK Version 1.4.365.23.
Description
One of our sample OPC UA Servers is updating a node value twice within the same sampling interval. Here is my test scenario -
Server has a minimum publish interval of 500ms and minimum sampling interval of 1000ms. I connect the server from UA Expert and create a subscription of a couple of nodes from my address space. I then change the value of the monitoring/sampling interval for one of the nodes from the client to 5000ms. The first node updates every second while the second sometimes updates twice every 5seconds. 
Can you please explain why this could be happening?

Environment :

OS: [Windows 10 pro]
Development environment: [Visual Studio 2019 16.7.2]
Runtime: [NET Core 3.1]
OPC Package Release Version [1.4.365.23]
Component: [Opc.Ua.Core,Opc.Ua.Server]
Server: [Reference Server]
Client: [UA Expert]

Praveen-g99, 2021-07-26:
Use case details - 
We have created an OPC UA Server using the .Net standard SDK and we have observed the below behavior.
Create a subscription with 500ms publish interval and add a monitored item with 5sec sampling interval. With this configuration from the client we see cases where the monitored item sometimes updates twice with a 1sec gap between them.
 For ex - the value updates at the 5th sec and 6th sec instead of skipping the 6th sec. It corrects it self again and repeats this behavior after a few more publish cycles.
upon further investigation we found the IncrementSampleTime API in Opc.Ua.Server/Subscription/MonitoredItem.cs. if we log the delta value we see that the value becomes negative at times and hence publishes the value again in the next publish cycle. Can you please have a look at this behavior and suggest what could be wrong

MD-V, 2021-07-27:
@Praveen-g99 

why are you using a PublishingInterval smaller than the SamplingInterval?

I think the best practice is to use a rather small SamplingInterval (e.g. 200ms) with a larger PublishingInterval (2 secs) and a matching QueueSize (e.g. 15).

See here: [https://www.prosysopc.com/blog/opc-ua-sessions-subscriptions-and-timeouts/](url)

mrsuciu, 2021-09-15:
Please re-open if this is still an issue

closed 2021-09-15 by mrsuciu