OPCFoundation/UA-.NETStandard #1446: Processing value of type ByteString

roland-braun, 2021-07-07:
**Type of Issue**
- [?] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [x] Question 
- [ ] Help wanted

**Describe the Issue**

The problem appeared when reading a value of a variable with a type definition DataTypeDictionaryType. The data type is ByteString. I was using UAExpert for testing the behavior of my OPC UA Server regarding custom data types. The address space of this server is created by importing a Nodeset file (PackML), which contains definitions of application-specific data types. Right after the UAExpert connects to this server, it starts importing the  meta data  of custom data types.  In one of these steps UAExpert reads the value of variable "TypeDictionary" defining an arbitrary IndexRange. The library implemented 
function: "public StatusCode ApplyRange(ref object value)" 
of class: public struct NumericRange
in namespce: Opc.Ua
responds with BadIndexRangeNoData. 
When reading the same variable using UAExpert's browse function with no IndexRange is applied, the read succeeds. 

While in a need for a solution I found a (quick) fix for function ApplyRange(...) like:

        public StatusCode ApplyRange(ref object value)
        {
            // check for empty range.
            if (this.m_begin == -1 && this.m_end == -1)
            {
                return StatusCodes.Good;
            }

            // nothing to do for null values.
            if (value == null)
            {
                return StatusCodes.Good;
            }

            Array array = value as Array;
            
            // 2021-07-06, Roland Braun 
            // Fix issue with reading (subsections from) ByteStrings
            if ( value is Variant vv )
            {
                if( vv.TypeInfo.BuiltInType.Equals(BuiltInType.ByteString))
                {
                    array = vv.Value as Array;
                }
            }

Remark: I have not tested the change regarding all possible situations. I was happy, once my application processed structured data types correctly. ;-)

Question 1: While data type "ByteString" is a scalar per definition (ValueRank = -1) how correct is the usage of an IndexRange?
Question 2: Is it correct or allowed to wrap a "ByteString" in an Opc.Ua.Variant when importing a related value from a Nodeset? 

**To Reproduce**
Steps to reproduce the behavior would need to transfer my application, which is not (yet) possible. 

**Expected behavior**
...

**Screenshots**
Read request (excerpt from wire shark)
[image]
Read response
[image]

**Log output**

**Code snippets**

**Environment (please complete the following information):**
 - OS: Windows 10
 - Development environment: Visual Studio 2019 16.10.3
 - Runtime: .NET 4.8
 - OPC Package Release Version [e.g. 1.4.363.107]
 - Component: Opc.Ua.Core
 - Server: my server
 - Client: UA Expert

**Additional context**
Add any other context about the problem here.

AlinMoldovean, 2021-07-12:
Hello @roland-braun ,

The ByteString and Byte[] array types are interchangeable. See #1425 
So a Read with IndexRange is valid for Node value.
It should be also possible to wrap it inside a Variant.

Could you please provide a small sample application to replicate the issue?

AlinMoldovean, 2021-07-16:
#1425

Suslsuk, 2021-07-19:
Roland, can you please provide us with the code of your UA server that you use to import the PackML Nodeset file and then display it in the namespace? I have not been able to do it so far with the information available here from similar issues (ImportXml etc.). 
That would certainly be a great help for many users who want to take advantage of the available nodeset.

Thanks

roland-braun, 2021-07-20:
> This email originated from outside of your organization. Please do not click on links or open attachments unless you recognize the sender and know the content is safe.

Hello Susluk

I need some time to extract the "simple example" from the product code.
I'm sorry if this causes any inconvenience but I need to squeeze the extra effort into my daily work.

Best regards
Roland

***@***.***D77D71.77ACF8C0]
-
Roland Braun
Networks & Devices
DECRC/S4

ABB AG Forschungszentrum
Wallstadter Straße 59
68526 Ladenburg
Germany
Phone: +49 6203 71-6417
E-Mail: ***@***.******@***.***>
www.abb.com<http://www.abb.com/>

ABB AG
Sitz/Head Office: Mannheim
Registergericht/Registry Court: Mannheim
Handelsregisternummer/Commercial Register No.: HRB 4664
Vorstand/Managing Board: Markus Ochsner (Vorsitzender/Chairman), Alexander Zumkeller
Vorsitzender des Aufsichtsrats/ Chairman of Supervisory Board: Klaus Eble

Diese E-mail enthält vertrauliche und/oder rechtlich geschützte Informationen. Wenn Sie nicht der richtige Adressat sind oder diese E-mail irrtümlich erhalten haben, informieren Sie bitte sofort den Absender und vernichten Sie diese Mail. Das unerlaubte Kopieren sowie die unbefugte Weitergabe dieser Mail ist nicht gestattet.
Bitte beachten Sie auch unsere Datenschutzerklärung für Geschäftspartner, die Sie über unsere Webseite<https://new.abb.com/privacy-policy/de/datenschutz> oder direkt unter diesem Link<https://new.abb.com/docs/librariesprovider30/default-document-library/datenschutzerkl%C3%A4rung-gesch%C3%A4ftspartner-abb-ag_de832794ecc1f463c09537ff0000433538.pdf> finden.

This E-mail may contain confidential and/or privileged information. If you are not the intended recipient (or have received this E-mail in error) please notify the sender immediately and destroy this E-mail. Any unauthorized copying, disclosure or distribution of the material in this E-mail is strictly forbidden.
Please also take note of our data protection notice for business partners, which you can find on our web page<https://new.abb.com/privacy-policy/de/datenschutz> or directly under the following link<https://new.abb.com/docs/librariesprovider30/default-document-library/data-protection-notice-business-partners-abb-ag_de762794ecc1f463c09537ff0000433538.pdf>.

roland-braun, 2021-07-23:
I tried sending a mail with the sample program. However the email was rejected becasue of its size. pls advise.

roland-braun, 2021-07-23:
Attaching the file using "drag&drop" did not work either. The response I get for the attemps says "is not included in the list". And after that, the comment where I tried to add a file, cannot be edited any more.

Suslsuk, 2021-07-26:
Here is a link to the github-comments max. filesizes & filetype. Hope that helps, thank you
https://docs.github.com/en/github/writing-on-github/working-with-advanced-formatting/attaching-files

mregen, 2021-09-15:
Hi @roland-braun, could you provide your sample code as a branch in a private fork? It appears difficult to send email with attachments via github.

EthanChangAED, 2023-07-07:
Closing this issue. Feel free to file a new one in Discussion-Q&A if there is more question.

closed 2023-07-07 by EthanChangAED