OPCFoundation/UA-.NETStandard #1447: Working with known types from the client

anlebr, 2021-07-07:
**Type of Issue**
- [ ] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [X] Question 
- [ ] Help wanted

Hi

I really appreciate the effort put in to make Opc Ua available open source!

### Background

I am working on a simple proof of concept for a visualization in Blazor.

I have a created a demo server representing an equipment that exposes a folder with two instances of "MeasurementType". The types are generated using the ModelCompiler. "Output" is of type "AnalogUnitRangeType"

[image]

I am using Blazor and the Nuget package to browse the server and dynamically generate a user interface. In this case that will be two components representing "Pressure" and "Humidity"

### Question

**When browsing a server and finding known self-defined types like MeasurementType. What is the easiest way to access the child nodes?**

I am browsing all nodes and checking if they are measurementTypes like this:

`if (node.TypeDefinitionId.Identifier.ToString() == TEM.Ahu.ObjectTypeIds.MeasurementType.Identifier.ToString())`

When I find the types I keep browsing the references and matching on displayName to find the variables I need.

Since I know the type and the ModelCompiler has generated a bunch of stuff, I am hoping that it might be possible to do something similar to this?

```
if(node is MeasurementState)
  {
      var measurementNode = node as MeasurementState;
  
      MeasurementContent content = new();
  
      content.DisplayName = measurementNode.DisplayName.Text;
      content.EngineeringUnit = measurementNode.Output.EngineeringUnits.Value.ToString();
      content.Value = (double)measurementNode.Output.Value;
      
  }

```

AlinMoldovean, 2021-07-12:
Hello @anlebr ,

Browsing the references and searching for a match in the results list is also an option.
But please make sure to search for a matching _BrowseName_ from type definition instead of _DisplayName_.

A more effective way of doing this is by using the _TranslateBrowsePathsToNodeIds_ service.
Please have a look at #1247 .

mrsuciu, 2021-09-15:
Please reopen if this still an issue

closed 2021-09-15 by mrsuciu

gdorigo, 2023-03-21:
Hi @anlebr, are you using blazor webassemply? if yes, do you have an example where you connect to opc server using blazor?