OPCFoundation/UA-.NETStandard #1449: Question

eoursel, 2021-07-07:
**Type of Issue**
- [ ] Bug 
- [X ] Enhancement 
- [ ] Compliance 
- [X ] Question 
- [ ] Help wanted

**Describe the Issue**

Why operations on CRL (IsRevoked, EnumerateCRL, DeleteCRL, AddCRL) are not supported on X509CertificateStore? Is there a workaround for these 4 methods by using CRLBuilder class ? 

Someone in my team has written an implementation which is based on Win32 API. Of course it will not run on Linux and Docker for sure so it is not the right solution. 

**Code snippets**

```
        /// <inheritdoc/>
        /// <remarks>CRLs are not supported here.</remarks>
        public bool SupportsCRLs => false;

        /// <inheritdoc/>
        /// <remarks>CRLs are not supported here.</remarks>
        public StatusCode IsRevoked(X509Certificate2 issuer, X509Certificate2 certificate)
        {
            return StatusCodes.BadNotSupported;
        }
```

Regards

mregen, 2021-07-15:
Hi @eoursel , the X509CertStore is based on the platform agnostic .NET X509Store implementation. To my knowledge it does not support CRL. That's why it is not supported.
My recommendation is to use the X509CertStore only for the application certificates, but Directory Store for trust lists.
The old .NET codebase also used win32 api, in order to support it many runtime checks need to be done and it doesn't work for docker/linux. Even worse, on macOS X509Store works only *very* limited.
There has been a proposed idea to make the cert store plugable in #1420 , would this be an option? Maybe you can help to develop the plug in, I asked the contributor for tests and a sample implementation (knowing it may not just be that easy) but it seems he had no time to contribute more. Before we can provide a solution the certificate valdator tests must pass. Its a security enhancement, so it requires special care.
Thanks,
Martin

eoursel, 2021-07-29:
hi @mregen, i understand the point, i need some time to check your proposal. While i think it would be nice to have a custom certificate store, i would prefer to stay with a standard solution. I agree that we need to be very careful with this very sensible source code. 
Best regards

mregen, 2021-09-15:
Hi @eoursel , closing this now as question appears to be answered.

closed 2021-09-15 by mregen