OPCFoundation/UA-.NETStandard #1458: Adding support for companion specs

joanna-zolopa, 2021-07-13:
I am quite new to OPC UA so forgive me if the question is rather basic. I
I am connecting to the OPC UA server that is using Machinery companion spec but am not quite sure what I need to do to get the DI types recognized on the client side. Is there a standard way to add support for companion specs to the reference client?

MD-V, 2021-07-13:
Hello @joanna-zolopa,

as there are no ModelCompiler compatible files for the Machinery companion spec you will need to decode/encode the types yourself. Check https://github.com/OPCFoundation/UA-.NETStandard/issues/1429#issuecomment-864378216 for more information.

MD-V, 2021-07-13:
Oh sorry you were talking about DI data types.

Check https://github.com/OPCFoundation/UA-Nodeset/tree/v1.04/DI for C# classes that you can use on client side.

closed 2021-09-15 by mregen