OPCFoundation/UA-.NETStandard #1461: MaxMessageCount is why doesnt work ?

fatihkutuk, 2021-07-14:
i am trying be a client to kepserverex opc server. in this client i have 61161 node (monitored item). when i try like under i have 
this exeption:  Could not send request to server.

AdditionalInfo = ">>> MaxMessageSize 4194240 < 4972068\r\n---    at Opc.Ua.BinaryEncoder.EncodeMessage(IEncodeable message)\r\n---    at Opc.Ua.BinaryEncoder.EncodeMessage(IEncodeable message, Stream stream, ServiceMessageContext context, Boolean leaveOpen)\r\n---    at Opc...

InnerResult = {BadEncodingLimitsExceeded 'MaxMessageSize 4194240 < 4972068'}

StatusCode = 2148007936

note : ( when i increase the number MaxMessageSize it dont working. When i limited 50000 node doesnt error and runnig quality)

code below
                _session = Session.Create(_config,
                new ConfiguredEndpoint(null, new 
                EndpointDescription(ConfigurationManager.ConnectionStrings["OpcStr"].ConnectionString)),
                true,
                "",
                360000,
                null,
                null).Result;
                
                _subscription = new Subscription(_session.DefaultSubscription)
                {
                    PublishingInterval = 2000,
                    MaxNotificationsPerPublish = 10000
       
                    
                };

                _subscription.AddItems(_nodes);

                Console.WriteLine(_subscription.MaxMessageCount);
                _subscription.FastDataChangeCallback += new FastDataChangeNotificationEventHandler(DataChanged);
                _session.AddSubscription(_subscription);
                _session.OperationTimeout = 3600000;
                _subscription.Create();

MD-V, 2021-07-15:
MaxMessageSize ist set on the server.

As far as I understand you are trying to add 61161  nodes to one subscription? Have you tried splitting it up to two subscriptions or adding the nodes to the subscription in bulks?

fatihkutuk, 2021-07-15:
> MaxMessageSize ist set on the server. Please check if the Kepware server supports adjusting it.
> 
> As far as I understand you are trying to add 61161 nodes to one subscription? Have you tried splitting it up to two subscriptions or adding the nodes to the subscription in bulks?

Your suggestion is very good, but the situation is a little different. There are 1400 devices in the field and there are 460000 nodes in total. I create a client according to the number of devices in the database and give the client numbers to the devices. In order to add new data to be read with Kepware rest api, I need to follow this path to manage which cilentta I will unsubscribe or subscribe to. Currently, when I create a client for every 110 device, some clients exceed 60000 objects, but when I make this number 100, there is no problem. but another problem is that when all devices are added, the number will increase a lot and kepware only allows 128 clients. So I wanted to increase the maxdatasize and see what I can do at most.

mregen, 2021-09-15:
Please reopen if there are still open questions..

fatihkutuk, 2021-09-15:
I contacted kepware, they reported that it can be fixed as max 4 mb in new versions and cannot be changed.

fatihkutuk, 2021-09-15:
I contacted kepware, they reported that it can be fixed as max 4 mb in a client in new versions and cannot be changed.

closed 2021-09-15 by fatihkutuk