OPCFoundation/UA-.NETStandard #1463: ctt and copy certificates to server

inspirit1986, 2021-07-19:
**Type of Issue**
- [ ] Question 

**Describe the Issue**

Please explain in what directory on reference server side I must copy contents of ctt - PKI\copyToServer\ApplicationInstance_PKI and \PKI\copyToServer\X509UserIdentity_PKI 

The chapter about ctt certs in Applications/ReferenceServer/README.md is apparently outdated:

> Placing the UACTT certificates
Copy the certificates for testing with the UACTT to the following stores:

> UA Applications/certs: expired.der, notyetvalid.der, opcuactt.der, opcuactt_incorrectappuri.der, opcuactt_incorrectip.der, opcuactt_incorrectsign.der, opcuactt_revoked.der, opcuser.der, opcuser_incorrectsign.der, opcuser_revoked.der, opcuser-expired.der, opcuser-notyetvalid.der, opcuactt_ca.der
UA Applications/crl: revocation_list_opcuactt_ca.crl

**Environment** 
- Windows 10 Pro En

**To Reproduce**
Steps to reproduce the behavior:
1. Install latest ctt 
2. Create server test project in ctt UI
3. ctt create folder within project with certs to copy - copytoserver/
4. start opcua reference server
5. reference server creates own cert folder in C:\ProgramData\OPC Foundation\pki\own

Thank in advance

inspirit1986, 2021-07-19:
I think I found the correct way.

after copy
copyToServer\ApplicationInstance_PKI\issuers\certs -> pki\issuer\certs
copyToServer\ApplicationInstance_PKI\issuers\crl -> pki\issuer\crl
copyToServer\ApplicationInstance_PKI\trusted\certs -> pki\trusted\certs
copyToServer\ApplicationInstance_PKI\trusted\crl -> pki\trusted\certs

copyToServer\X509UserIdentity_PKI\issuers\certs -> pki\issuerUser\certs
copyToServer\X509UserIdentity_PKI\issuers\crl ->  pki\issuerUser\crl
copyToServer\X509UserIdentity_PKI\trusted\certs -> pki\trustedUser\certs
copyToServer\X509UserIdentity_PKI\trusted\crl -> pki\trustedUser\crl

ctt stop warning on security facet. 
Have I understood everything correctly?

AlinMoldovean, 2021-07-19:
Hello @inspirit1986 ,

Yes, your procedure for copying the CTT certificates to Server's PKI is correct.
Make sure to also copy the Server certificate to CTT trust list, i.e.  to "CTT project folder\PKI\CA\certs"

mregen, 2021-09-15:
Please reopen if there are still open questions

closed 2021-09-15 by mregen