OPCFoundation/UA-.NETStandard #1467: Dealing with Extension Objects

szymonlechowicz, 2021-07-22:
**Type of Issue**

- [x] Help wanted

**Describe the Issue**

I'm making OPC UA Client. As I said in earlier issue, this is completely new to me. 
I want this program to call methods like on the example below (from Siemens Documentation)

[image]

Problem is that program sees input or output arguments of any method as Extension Object. I don't how to deal with that. I'm searching information about reading Extension Objects via Internet but all I can find is information it is weakly supported. 

**Environment (please complete the following information):**
- OS: Windows 10
- Development environment: [Visual Studio 2019 16.9.0]
- Runtime: [.NET Framework 4.7.2]
- OPC Package Release Version [available on Nuget]

mrsuciu, 2021-07-29:
@szymonlechowicz 
The input argument is an array of ExtensionObject and the Body property of each contained ExtensionObject is of type Argument.
The Argument contains DataType and ValueRank properties that define the requested types for the arguments.

mregen, 2022-01-21:
Please reopen if necessary...

closed 2022-01-21 by mregen