OPCFoundation/UA-.NETStandard #1468: Client isn't running.

PraveenKumarSoni, 2021-07-22:
**Type of Issue**
- [ ] Help wanted

**Describe the Issue**
When I'm trying to run client then, I'm getting errors. 

**Expected behavior**
A client should run in a normal way. 

**Environment (please complete the following information):**
 - OS: Ubuntu
 - Development environment: [e.g. Visual Studio 2019 16.7.2]
 - Runtime: .NET Core 5.0 and 3.1
 - OPC Package Release Version [e.g. 1.4.363.107]
 - Server: [e.g. Reference Server]
 - Client: [e.g. UA 

**Here is the error:-**
`/usr/share/dotnet/sdk/5.0.302/Microsoft.Common.CurrentVersion.targets(2202,5): warning MSB3243: No way to resolve conflict between "Opc.Ua.Client, Version=1.4.366.0, Culture=neutral, PublicKeyToken=bfa7a73c5cf4b6e8" and "Opc.Ua.Client, Version=1.4.0.0, Culture=neutral, PublicKeyToken=null". Choosing "Opc.Ua.Client, Version=1.4.366.0, Culture=neutral, PublicKeyToken=bfa7a73c5cf4b6e8" arbitrarily. [/home/opcuaCSharp/UA-.NETStandard/Applications/ConsoleReferenceClient/ConsoleReferenceClient.csproj]
/usr/share/dotnet/sdk/5.0.302/Microsoft.Common.CurrentVersion.targets(2202,5): warning MSB3243: No way to resolve conflict between "Opc.Ua.Client, Version=1.4.366.0, Culture=neutral, PublicKeyToken=bfa7a73c5cf4b6e8" and "Opc.Ua.Client, Version=1.4.0.0, Culture=neutral, PublicKeyToken=null". Choosing "Opc.Ua.Client, Version=1.4.366.0, Culture=neutral, PublicKeyToken=bfa7a73c5cf4b6e8" arbitrarily. [/home/opcuaCSharp/UA-.NETStandard/Applications/ConsoleReferenceClient/ConsoleReferenceClient.csproj]
/home/opcuaCSharp/UA-.NETStandard/Applications/ConsoleReferenceClient/UAClient.cs(61,16): error CS0433: The type 'Session' exists in both 'Opc.Ua.Client, Version=1.4.366.0, Culture=neutral, PublicKeyToken=bfa7a73c5cf4b6e8' and 'Opc.Ua.Client, Version=1.4.0.0, Culture=neutral, PublicKeyToken=null' [/home/opcuaCSharp/UA-.NETStandard/Applications/ConsoleReferenceClient/ConsoleReferenceClient.csproj]
/home/opcuaCSharp/UA-.NETStandard/Applications/ConsoleReferenceClient/UAClient.cs(446,50): error CS0433: The type 'MonitoredItem' exists in both 'Opc.Ua.Client, Version=1.4.366.0, Culture=neutral, PublicKeyToken=bfa7a73c5cf4b6e8' and 'Opc.Ua.Client, Version=1.4.0.0, Culture=neutral, PublicKeyToken=null' [/home/opcuaCSharp/UA-.NETStandard/Applications/ConsoleReferenceClient/ConsoleReferenceClient.csproj]
/home/opcuaCSharp/UA-.NETStandard/Applications/ConsoleReferenceClient/UAClient.cs(446,79): error CS0433: The type 'MonitoredItemNotificationEventArgs' exists in both 'Opc.Ua.Client, Version=1.4.366.0, Culture=neutral, PublicKeyToken=bfa7a73c5cf4b6e8' and 'Opc.Ua.Client, Version=1.4.0.0, Culture=neutral, PublicKeyToken=null' [/home/opcuaCSharp/UA-.NETStandard/Applications/ConsoleReferenceClient/ConsoleReferenceClient.csproj]
/home/opcuaCSharp/UA-.NETStandard/Applications/ConsoleReferenceClient/UAClient.cs(494,17): error CS0433: The type 'Session' exists in both 'Opc.Ua.Client, Version=1.4.366.0, Culture=neutral, PublicKeyToken=bfa7a73c5cf4b6e8' and 'Opc.Ua.Client, Version=1.4.0.0, Culture=neutral, PublicKeyToken=null' [/home/opcuaCSharp/UA-.NETStandard/Applications/ConsoleReferenceClient/ConsoleReferenceClient.csproj]
/home/opcuaCSharp/UA-.NETStandard/Applications/ConsoleReferenceClient/UAClient.cs(61,16): error CS0433: The type 'Session' exists in both 'Opc.Ua.Client, Version=1.4.366.0, Culture=neutral, PublicKeyToken=bfa7a73c5cf4b6e8' and 'Opc.Ua.Client, Version=1.4.0.0, Culture=neutral, PublicKeyToken=null' [/home/opcuaCSharp/UA-.NETStandard/Applications/ConsoleReferenceClient/ConsoleReferenceClient.csproj]
/home/opcuaCSharp/UA-.NETStandard/Applications/ConsoleReferenceClient/UAClient.cs(446,50): error CS0433: The type 'MonitoredItem' exists in both 'Opc.Ua.Client, Version=1.4.366.0, Culture=neutral, PublicKeyToken=bfa7a73c5cf4b6e8' and 'Opc.Ua.Client, Version=1.4.0.0, Culture=neutral, PublicKeyToken=null' [/home/opcuaCSharp/UA-.NETStandard/Applications/ConsoleReferenceClient/ConsoleReferenceClient.csproj]
/home/opcuaCSharp/UA-.NETStandard/Applications/ConsoleReferenceClient/UAClient.cs(446,79): error CS0433: The type 'MonitoredItemNotificationEventArgs' exists in both 'Opc.Ua.Client, Version=1.4.366.0, Culture=neutral, PublicKeyToken=bfa7a73c5cf4b6e8' and 'Opc.Ua.Client, Version=1.4.0.0, Culture=neutral, PublicKeyToken=null' [/home/opcuaCSharp/UA-.NETStandard/Applications/ConsoleReferenceClient/ConsoleReferenceClient.csproj]
/home/opcuaCSharp/UA-.NETStandard/Applications/ConsoleReferenceClient/UAClient.cs(494,17): error CS0433: The type 'Session' exists in both 'Opc.Ua.Client, Version=1.4.366.0, Culture=neutral, PublicKeyToken=bfa7a73c5cf4b6e8' and 'Opc.Ua.Client, Version=1.4.0.0, Culture=neutral, PublicKeyToken=null' [/home/akshata/opcuaCSharp/UA-.NETStandard/Applications/ConsoleReferenceClient/ConsoleReferenceClient.csproj]`

mrsuciu, 2021-07-28:
@PraveenKumarSoni
Could it be the case that you also referred the nuget package OPCFoundation.NetStandard.Opc.Ua under the ConsoleReferenceClient project dependencies and at the same time refer under the Projects dependencies the Opc.Ua.Client project?
You can check it under VS 2019 by expanding **Dependencies/netcoreapp3.1/Packages" and **Dependencies/netcoreapp3.1/Projects" from the ConsoleReferenceClient project.
This would produce such an error at compile time.
If this is the case you should exclusively either use the nuget package or the Projects dependencies.

mregen, 2021-09-15:
Please reopen if there are still questions

closed 2021-09-15 by mregen