OPCFoundation/UA-.NETStandard #1470: Get nodeId and value in same object when reading from nodes?

rafaelschlatter, 2021-07-23:
**Type of Issue**
- [x] Question 

I am looking at the `ConsoleReferenceClient` here: https://github.com/OPCFoundation/UA-.NETStandard/blob/master/Applications/ConsoleReferenceClient/UAClient.cs

On lines 170-178 the client sets up the Value and BrowseName of `ServerStatus_StartTime` to be read later on in `m_session.Read()`. That method returns objects of type `DataValue` and if I want to link the Value of `ServerStatus_StartTime` to its BrowseName I track that myself (the only way to do this that I see is using the order from `ReadValueIdCollection`).

Is there a way to read nodes and get an object that contains both the value and the BrowseName (or better the nodeId)?

AlinMoldovean, 2021-07-23:
Hello @rafaelschlatter ,

You can use the `Session.ReadNode(NodeId nodeId)` method https://github.com/OPCFoundation/UA-.NETStandard/blob/98634b1cc71bcd59bc477f9146bec1106b7b80d5/Libraries/Opc.Ua.Client/Session.cs#L1618

It returns a _Node_ object after reading the complete set of attributes for a _NodeId_

rafaelschlatter, 2021-07-23:
Hi @AlinMoldovean

Thanks for the swift reply. How do I get the value after I read the node? I can see the nodeId and browseName attributes, but I am not sure how to get the value for the node I just read? The value is not among the attributes on lines 1621-1648 in `Session.cs` that you linked.

Is `ReadNode()` the right method for my problem?

[image]

AlinMoldovean, 2021-07-26:
Hi @rafaelschlatter ,

It seems that _Value_ attribute is not read for _Variable_ nodes.
I think that's because it is dynamic by design and can change frequently.
Anyway, you can read and set  it like this:

```
                // Read Node
                Node node = m_session.ReadNode(Variables.Server_ServerStatus);

                if (node is VariableNode variableNode)
                {
                    variableNode.Value = m_session.ReadValue(Variables.Server_ServerStatus);
                }
```

mregen, 2021-09-15:
Please reopen if there are still questions open

closed 2021-09-15 by mregen

rafaelschlatter, 2021-09-15:
No, sorry for the too late reply. Thank you @AlinMoldovean for the help.