OPCFoundation/UA-.NETStandard #1475: X509 CRL Distributions Points

eoursel, 2021-07-29:
**Type of Issue**
- [X] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [ ] Question 
- [ ] Help wanted

**Describe the Issue**

It seems that a list of distribution points is not accepted as a X509Extension. Only a single distribution point is allowed in the parameter of X509.BuildX509CRLDistributionsPoints

**To Reproduce**
In unit test CreateCACertForRSA, the default usage of AddExtension is the following

```
   var cert = CertificateBuilder.Create(Subject)
                .SetCAConstraint(-1)
                .SetHashAlgorithm(keyHashPair.HashAlgorithmName)
                .AddExtension(X509Extensions.BuildX509CRLDistributionPoints("http://myca/mycert.crl"))
                .SetRSAKeySize(keyHashPair.KeySize)
                .CreateForRSA();

```
**Expected behavior**
We should be able to write something like that

```
   var cert = CertificateBuilder.Create(Subject)
                .SetCAConstraint(-1)
                .SetHashAlgorithm(keyHashPair.HashAlgorithmName)
                .AddExtension(X509Extensions.BuildX509CRLDistributionPoints("http://myca/mycert.crl","http://myca2/mycert.crl"))
                .SetRSAKeySize(keyHashPair.KeySize)
                .CreateForRSA();

```

**Environment (please complete the following information):**

 - Runtime: .NET Core 3.1
 - OPC Package Release 1.4.366.38
 - Component: [e.g. Opc.Ua.Security.Certificates]

mregen, 2021-07-30:
@eoursel, agreed...

mregen, 2021-08-23:
Hi @eoursel , currently the API supports only a hightly simplified version of the distribution point.
The 'proper' fix were to support the RFC:

```
id-ce-cRLDistributionPoints OBJECT IDENTIFIER ::=  { id-ce 31 }

   CRLDistributionPoints ::= SEQUENCE SIZE (1..MAX) OF DistributionPoint

   DistributionPoint ::= SEQUENCE {
        distributionPoint       [0]     DistributionPointName OPTIONAL,
        reasons                 [1]     ReasonFlags OPTIONAL,
        cRLIssuer               [2]     GeneralNames OPTIONAL }

   DistributionPointName ::= CHOICE {
        fullName                [0]     GeneralNames,
        nameRelativeToCRLIssuer [1]     RelativeDistinguishedName }

   ReasonFlags ::= BIT STRING {
        unused                  (0),
        keyCompromise           (1),
        cACompromise            (2),
        affiliationChanged      (3),
        superseded              (4),
        cessationOfOperation    (5),
        certificateHold         (6),
        privilegeWithdrawn      (7),
        aACompromise            (8) }
```

Do you need the full support? Basically the same issue with AIA extension.

mregen, 2021-09-04:
Hi @symarchand, the fix looks good to me, to get the credit for it and to sign the contributor agreement I suggest you open a PR please?

PR #1504 (2021-09-13): add API for multiple distribution points
fixes #1475
commit dad6bcd8 (2021-09-13): Add API for multiple CRL distribution points (#1504)

fixes #1475
closed 2021-09-13 by mregen