OPCFoundation/UA-.NETStandard #1476: .Net type for UD Struct is not being emitted by the ComplexTypeSystem

hermoter, 2021-08-02:
**Type of Issue**
- [ ] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [ X ] Question 
- [ X ] Help wanted

**Describe the Issue**

For integration test purposes (in 2020), I created a Docker image with a [NetCoreConsoleServer](https://github.com/OPCFoundation/UA-.NETStandard-Samples/tree/master/Samples/NetCoreConsoleServer) and a customized [TestDataDesign.yml](https://github.com/OPCFoundation/UA-.NETStandard-Samples/blob/master/Samples/Opc.Ua.Sample/TestData/TestDataDesign.xml) model that was compiled with the [UA-ModelCompiler](https://github.com/OPCFoundation/UA-ModelCompiler).
This custom model includes my own UD structure data type (that inherits from Structure datatype) with 4 fields.
I make use of the [ComplexTypeSystem's](https://github.com/OPCFoundation/UA-.NETStandard/tree/master/Libraries/Opc.Ua.Client.ComplexTypes) emitted .Net types to handle encoding and decoding of UD structs when reading / writing to nodes of this type.

All worked well until I recently had to make small changes to my UD struct.
I don't have the same UA-Compiler that I used in 2020, so I had to pull the latest of the UA-ModelCompiler in order to compile the modified model again.
So after compiling the model with a more recent UA-ModelCompiler, reading and writing of nodes of this struct type do not work as expected.
On closer inspection, I found that the ComplexTypeSystem does not return the struct type definition if any of the fields are optional (IsOptional == true) as seen [here](https://github.com/OPCFoundation/UA-.NETStandard/blob/b5421ea39161fac58df5b3ae5179230384110177/Libraries/Opc.Ua.Client.ComplexTypes/ComplexTypeSystem.cs#L691)
I then checked in UAExpert to see if there are differences between the model from 2020 and the more recently compiled model, and I discovered that the model from 2020 shows all fields as required (IsOptional == false), whereas for the recently compiled model, all the fields are optional. (See screenshots)
So something must have changed since the previous version of the ModelCompiler that I used in 2020.

I then tried to get the UA-ModelCompiler to compile the fields as required (IsOptional == false), but I haven't managed to get it working. It seems to always compile struct fields as being optional.
On further inspection, I also found that the IsOptional property in the Opc.Ua.DataTypes.StructureField class is true by default as seen in the [StructureField.Initialize](https://github.com/OPCFoundation/UA-.NETStandard/blob/master/Stack/Opc.Ua.Core/Stack/Generated/Opc.Ua.DataTypes.cs) method.
This property only gets set based on a few conditions in [ModelDesignValidator.GetStructureDefinitionFields](https://github.com/OPCFoundation/UA-ModelCompiler/blob/753c8c88934df06a14f11bfa1e3899caadbf7e51/ModelCompiler/ModelDesignerValidator.cs#L6202) that are not clear to me.

It isn't clear to me how a struct's fields can be specified as either optional or required at compile time.
Furthermore, it is unclear why the ComplexTypeSystem does not return a struct's definition if any of the fields are optional.
I am trying to understand the reason between support for structs with required fields in the ComplexTypeSystem vs no support for structs with at least one optional field.

Can anybody please
 1) indicate what I am missing with regards to defining a struct's fields as optional vs required,
 2) explain why the ComplexTypeSystem does not seem to support structs with optional fields
 3) point me to any specification explaining the rules regarding structs with optional fields vs required fields.

**To Reproduce**
Steps to reproduce the behavior:
Issue description above gives a clear idea of the steps.

**Expected behavior**
I would expect to see the ComplexTypeSystem work for any kind of structs, regardless of optional vs required fields, unless there is some clear documentation that explains the rules and different scenarios.
I would expect my struct's fields to be required by default, once compiled with the UA-ModelCompiler, unless there is a reason why the default was changed from IsOptional == false to IsOptional == true.
I would expect to see a .Net type emitted for my type, regardless of optional or required fields.

**Screenshots**
This is what I see in UAExpert when I look at the struct field definition:
First one is original model from 2020
Second is the recently compiled model showing fields as optional (and causing the ComplexTypeSystem to not work as expected)
[image]

[image]

**Log output**

**Code snippets**

**Environment (please complete the following information):**

**Additional context**

mregen, 2021-08-19:
Hi @hermoter , could you share your ModelCompiler input/output with differences in the new ModelCompiler output?

mregen, 2021-08-19:
Hi @hermoter , this condition: [here](https://github.com/OPCFoundation/UA-.NETStandard/blob/b5421ea39161fac58df5b3ae5179230384110177/Libraries/Opc.Ua.Client.ComplexTypes/ComplexTypeSystem.cs#L691) indicates a false combination of struct and optional fields is used. The type should be StructureType.StructureWithOptionalFields.

I suspect you are seeing an issue with: Modelcompiler updates, errata 1.04.9 nodeset (#1284)
The way how the datatypedefinitions and structures are built during load was changed. So you need to make sure, if you used the latest modelcompiler to use also a latest nuget package to import the model. Please check in UANodesetHelper / Import if your model is properly imported.

mregen, 2021-08-19:
Hi @hermoter , I can repro the issue here. I will check with the owner of the ModelCompiler if the change was intentional.
Maybe the check in the complex type loader for IsOptional should be removed..
Thanks, Martin

PR #1503 (2021-09-13): Complex types: Ignore IsOptional fields for structure
fixes #1476 
some ModelCompiler versions produce invalid output, for Structure it can be ignored
commit a4bf0ab1 (2021-09-13): Complex types client can ignore isoptional fields for structure (#1503)

fixes #1476 
 Improve IOP, some recent ModelCompiler versions (starting with errata 1.04.9) produce invalid output and all IsOptional fields have become true by default, causing the complex types client to bail out for Structures created with this ModelCompiler release. However, the IsOptional field can be ignored.
ModelCompiler is fixed in 1.04.10 and greater
closed 2021-09-13 by mregen