OPCFoundation/UA-.NETStandard #1477: Issues building the reference server in VS 2019

om327, 2021-08-05:
**Describe the Issue**

Hello, I'm not a .NET developer and tend to us LabVIEW for development and system integration. I really like the flexability of OPC UA and have developed several labVIEW OPC UA servers using the LabVIEW OPC UA Toolkit. I'd like to create OPC UA servers using the offical .NET stack since the LabVIEW implementation doesnt support methods. I'm trying to follow the instructions for building and running the reference server from the .NET OPC UA Stack but can not seem to load and build it in VS 2019? I've also tried following the GitHub server development tutorial (https://opcfoundation.github.io/UA-.NETStandard/help/index.htm#server_development.htm) but it appears to be out of date (last updated 2015) and the instructions no longer match the folders for the latest builds from GitHub. The closest i've gotten is to load the pacakages into VS but it then shows reference errors for the Opc.Ua.xxxx.controls?

I've installed VS 2019 and .NET 4.6.2 SDK but I have 241 errors? Are there any detailed instructions for what specfically needs to be installed and how to build the reference server or updates to the docuemnts for building your own server?

**To Reproduce**
Steps to reproduce the behavior:
clone OPC UA .NETStandard into VS 2019 Repo
double click on UA Reference.sln  

**Screenshots**
[image]

**Environment (please complete the following information):**
Windows 10
Visual Studio 2019

Thanks for the help!
Ollie

MD-V, 2021-08-06:
Hello @om327,

the errors show that your nuget source ("Microsoft Visual Studio Offline Packages") is missing many nuget packages. try adding nuget.org to your nuget sources (See [](https://www.grapecity.com/componentone/docs/uwp/online-getting-started/config-nugetpackage.html)) or manually download all needed packages and add them to a local source.

om327, 2021-08-06:
That did the trick - I thought it would be something daft that i didn't do! Thanks for the help!

closed 2021-08-06 by om327