OPCFoundation/UA-.NETStandard #1481: Disable check for data-encipherment flag

einarmo, 2021-08-16:
**Type of Issue**
- [ ] Bug 
- [ x] Enhancement 
- [ ] Compliance 
- [ ] Question 
- [ ] Help wanted

**Describe the Issue**

There is currently no way to work around servers with certificates that lack the data-encipherment bit, as seen [here](https://github.com/OPCFoundation/UA-.NETStandard/blob/7e736d22fc2e3109f4e5bd9d2a8a637a54a34bf8/Stack/Opc.Ua.Core/Security/Certificates/CertificateValidator.cs#L933). While servers that lack this bit are in violation of the standard (?), it would be nice to have some way to work around it, even if that means intentionally lowering security.

**To Reproduce**
Connect to a server that uses a certificate that lacks the data-encipherment flag. The client should produce BadCertificateUseNotAllowed.

**Expected behavior**
This _is_ expected behavior. But it would be nice to have an option to disable this check, like with SHA1 signed certs. For now I'm stuck having to convince the owners of the server that the fault is on their end, and that the reason why their client application works and not mine is because they technically have a bug in their client. It is nice to have a way to work around issues like this when developing client applications, since making the fix on the server can be hard or impossible.

I can make a PR to fix this myself, if you agree that a config option is reasonable.

einarmo, 2021-08-19:
Turns out this can be solved using the certificate callback, it might be worth considering, but it isn't as important as I initially thought.

closed 2021-08-19 by einarmo

mregen, 2021-08-20:
jfyi the cert callback only allows to accept suppressible errors.

einarmo, 2021-08-20:
For reference, if anyone else runs into similar problems, [here is the list of suppressible certificate errors](https://github.com/OPCFoundation/UA-.NETStandard/blob/7e736d22fc2e3109f4e5bd9d2a8a637a54a34bf8/Stack/Opc.Ua.Core/Security/Certificates/CertificateValidator.cs#L1178).