OPCFoundation/UA-.NETStandard #1483: Why one network interface is not enough for UDP message deliver?

aehomay, 2021-08-16:
**Type of Issue**
- [ ] Bug 
- [ ] Enhancement 
- [ ] Compliance 
- [X] Question 
- [ ] Help wanted

Why does the **GetUdpClients** method need to return a list of UdpClient? Isn`t one interface enough to broadcast UDP messages? 

Line number: 107
Libraries/Opc.Ua.PubSub/Transport/UdpClientCreator.cs

mrsuciu, 2021-08-20:
Since the network address one can specify, might be a limited/shared broadcast address or multicast address which depending on the network configuration might span on multiple network interfaces, publishing UDP should be done through all of them.

mrsuciu, 2021-09-08:
@aehomay Probably I don't quite get the idea but as I see CreateUdpClientForNetworkInterface method does that when necessary before adding the UdpClient to the returned list.

mregen, 2022-01-21:
Consider conversation resolved.

closed 2022-01-21 by mregen