OPCFoundation/UA-.NETStandard #1485: Retrieve the state of the disconnected channel in the Opc.Ua.Client.cs class.

Perpete, 2021-08-16:
Hello,
Following a problem encountered and waiting for help to write or read from a node with the network cable disconnected during the keep Alive late phase (See # 1373), I am trying to find a solution.

I was thinking of blocking the action of the Opc.UA.Client.cs class read and write service call functions when the cable was unplugged.
For a test, I would like to retrieve the state of the disconnected channel in the Opc.Ua.Client.cs class.
Is it possible?
Where to find this information?
Or any other solutions to resolve problem # 1373.

mrsuciu, 2021-08-19:
@Perpete 
Hi,
I've looked into thread #1373. 
I think the situation occurs due to the fact that you are executing the reads/writes from the main thread and when you have unplugged the cable, most probably because the PLC hasn't answered to all the read/write requests send by the client, you will have an outstanding request pending that keeps the main thread blocked.
The main thread being blocked is blocking in turn all the callbacks/event handlers that check the `if (this.InvokeRequired)` statement. They are queued but never executed until the outstanding request gets completed or times out.
`Session_KeepAlive` is such an event handler.
Your print screen depicts clearly this situation:
https://user-images.githubusercontent.com/66659162/128130435-05586beb-20c4-4818-b0fb-f87e40093f7f.png

The solution is to call the read/writes from a different thread other than the main thread, or use their asynchronous versions.

Perpete, 2021-08-23:
Hello mrsuciu,
Thank you for your information.
Can you point me where I could find an example of using the asynchronous version of reading and writing a node?
I can't find this use in UA-.NETStandard-Samples-master nor in SimpleOpcClient-master.

Perpete, 2021-08-24:
Hello,
While searching the net for a long time, I found an example of using the asynchronous read and write version of a node in the OpcUaHelper-master project on Github with the functions **Mysession.BeginRead** and **Mysession. BeginWrite**..
I applied the asynchronous solution.
Now, I no longer have a problem with locking the main thread when reading or writing a node when the network cable is unplugged just before the late keep Alive phase.
Thanks for your help.

closed 2021-08-24 by Perpete