OPCFoundation/UA-.NETStandard #1491: CTT issue : The tool seems to not encrypt when the OPCUA server certificate is build with a key lenght equals to 4096 is used

GregoireG-C, 2021-08-20:
**Type of Issue**
- [ ] Bug 
- [ ] Enhancement 
- [x] Compliance 
- [ ] Question 
- [X] Help wanted

**Describe the Issue**

In CTT, the tests in Security User Name Password failed when I use a certificate with a key lenght of 4096 bits while it works when I use a certificate with a key lenght of 3072 bits. When I watch closely in the CTT debuger, it seems that the CTT is unable to encrypt the message with a key lenght of 4096.

**To Reproduce**
Steps to reproduce the behavior:
1.Create a certificate with the following settings :
 - sign Algorythm : sha256RSA
 - Hach sign algo : sha256
 - public key lenght 4096 bits
2. Configure your OPCUA server to use this certificate
3. Launch the server and launch the conformance unit : Security/Security UserName Password
4. Check that initialize and 001.js has failed

**Expected behavior**
The initialize and 001.js succeed

**Screenshots**

The first screenshot is obtained when using a keyLenght of 4096 while the second one is obtained with a keyLenght of 3072.
You can observe that the data in the debugger is on the first screenshot null and on the second one completed.
[image]
[image]

CTT version : 1.4.9.398

mregen, 2021-08-31:
Thanks @GregoireG-C ,
the issue has been reported to the OPCF compliance team tracked by: https://mantis.opcfoundation.org/view.php?id=7209

GregoireG-C, 2021-09-08:
Hi mregen

I reported an other bug on the OPC compliance team mantis:

Precisions : I tested only with the CTT version 1.02

When the server certificate  subject does not contain the DC (domainComponent), the CTT can't encrypt the message. I didn't debuggedd to see why.

According to The OPCUA norm table n°3. OPCUA follow the rfc3280 norm.
Exctracted from rcf3280 : 

> 5.2.2  Issuer Alternative Name
> 
>    The issuer alternative names extension allows additional identities
>    to be associated with the issuer of the CRL.  Defined options include
>    an rfc822 name (electronic mail address), a DNS name, an IP address,
>    and a URI.  Multiple instances of a name and multiple name forms may
>    be included.  Whenever such identities are used, the issuer
>    alternative name extension MUST be used; however, a DNS name MAY be
>    represented in the issuer field using the domainComponent attribute
>    as described in section 4.1.2.4.

Our server certificate specify its DNS into the the issuer alternative names extension and do not specify the DC into the subject (which is complient with the rcf norm). This makes the test fail.

mregen, 2022-01-21:
> Hi mregen
>
> I reported an other bug on the OPC compliance team mantis:
>
> Precisions : I tested only with the CTT version 1.02
>
> When the server certificate subject does not contain the DC [...]

Hi @GregoireG-C , I noticed the Mantis issue is closed, is the latter issue being addressed? I think the content of the certificates at first needs to follow Opc Ua spec and the latest Rfc5280. Rfc 3280 became obsolete. Please reopen a new issue if the topic is still of interest.

mregen, 2022-01-21:
Mantis closed, next CTT release contains the fix

closed 2022-01-21 by mregen